MarlinFirmware/Marlin #16685: [BUG] Set Home Offsets from LCD - KILLED/Err

onkytonk, 2020-01-26:
### Bug Description

Running SKR e3 Mini v1.2 on an Ender 3 Pro with a BLTouch v3.1
After homing, if I try to adjust my home offsets using the LCD, its results in the printer being Halted/KILLED. 

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4114728/Marlin.zip)

### Steps to Reproduce

1) Home Printer
2) Select Menu > Configuration > Advanced Settings > Set Home Offsets

If I reset EEPROM, and repeat above, I get "Err: Too far!"

**Expected behavior:** [What you expect to happen]
I should be able to adjust my home offsets

**Actual behavior:** [What actually happens]
Printer is halted and I'm instructed to reset. Or I received "Err: Too far"

boelle, 2020-01-28:
is this with bugfix 2.0.x?

Vertabreak, 2020-01-31:
home offsets do not function in such a way that they can be used with 3dprinting without additional editing to make it do so its part of the cnc system seems to be working as intended.

sjasonsmith, 2020-02-01:
> If I reset EEPROM, and repeat above, I get "Err: Too far!"

I can reproduce this, although I didn't even know this option existed so I don't know what is the proper behavior.

sjasonsmith, 2020-02-01:
It is worth reading this description of the behavior:
https://marlinfw.org/docs/gcode/M428.html

Based on this, I think it is behaving as expected. I agree with VertaBreaker that it probably doesn't serve much purpose for most 3D printer users.

This will be disabled if `NO_WORKSPACE_OFFSETS` is enabled. Perhaps that should be the default for examples and the default configurations.

sjasonsmith, 2020-02-01:
Of course anything reproducible that results in KILLED shouldn't be expected behavior. I was referring to the "Too Far" behavior.

thinkyhead, 2020-02-01:
@onkytonk — `M428` doesn't do very much. If it's causing a KILL that is strange. So here's a strange thing to try: See if it helps to enable the `SPEAKER` option.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle