MarlinFirmware/Marlin #16686: [BUG] UBL - Y Jumping to half bed size after a while

zmiguel, 2020-01-27:
### Bug Description

When Running `G29 P1` after a `M502` & `G28` UBL starts to generate the mesh as expected, however, after some time the Y value is set to 150 (half my bed size) while the nozzle is NOT at 150.

I've tried with point per axis at 9, 10 and 11. On 11 it gets to around 82 points and Y position is incorrect, and on 9 it happens at around 60, so doesn't seem to be a fixed value issue

### My Configurations

SKR 1.4 Turbo Board + 3DTouch (BLTouch Clone)
300x300 bed size
20mm inset for glass clips

[Gist with Config](https://gist.github.com/zmiguel/54cafdfb3da530c52fde6f5e6a49e799)

### Steps to Reproduce

1. `M502` + `G28` + `G29 P1`
2. Wait

**Expected behavior:** 

Marlin should generate the UBL mesh.

**Actual behavior:**

Marlin starts generating the bed and fails near the end by incorrectly reporting Y position making the mesh generation fail because the nozzle/probe assembly tried to go out of the bed.

#### Additional Information

Nothin I can think of, but I had this working once. Now (same marlin version) it fails every time, Bilinear and other methods work fine.

boelle, 2020-01-28:
remember to do a M500 after M502

zmiguel, 2020-01-29:
@boelle Tried that too, didn't help. Keeps jumping at around 3/4 of the way through the end

boelle, 2020-02-08:
@zmiguel have you tried to retest with bugfix 2.0.x (the latest copy)

RuairiSpain, 2020-02-10:
Your MESH_INSET is 10 but your probe offset is:`#define NOZZLE_TO_PROBE_OFFSET { 39, 0, -3.15 }`

Try increasing your MESH_INSET to 40 (to take account of the 39mm for the your probe x offsets) and re-generate mesh.  UBL creates the mesh coordinates from the home outwards in circles, once it hits a unreachable position it stops and you need to manually edit the mesh.  If you increase the mesh inset, that should avoid unreachable points.  If you want more control of the mesh min/max rather then MESH_INSET, have a look at configuration_adv.h, there are 4 setting to change the mesh left/right/front/back

ManuelMcLure, 2020-02-10:
The probe offsets are already taken into account when determining where to probe. You don't have to add the probe offsets to ``MESH_INSET`` - doing this will just reduce the area that can corrected.

When you probe the bed the _probe_ will go to the various points of the bed, so you will not see the X and Y on the LCD match the position of the probed point but rather the position of the nozzle when the point is probed. So, for example, when probing the center of the bed the _probe_ will be at 150, 150 and the X and Y will show up as 111, 150 (which is the position of the _nozzle_).

It is _normal_ that not all points can be probed. UBL is designed to be able to have a mesh that is larger than the area that can be probed. Once you have completed the ``G29 P1`` you can use ``G29 P3`` (possibly several timed) to auto-fill the points that could not be probed. Then you can use ``G29 P4`` or ``M421`` to fine-tune the points that were not probed.

thinkyhead, 2020-02-11:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

closed 2020-02-16 by thinkyhead