MarlinFirmware/Marlin #16691: [BUG] Can't compile Marlin 2.0.2

qwewer0, 2020-01-27:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4117195/Configuration.zip)

**VS Code terminal output:**

```
In file included from Marlin\src\module/stepper.h:47:0,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/indirection.h:701:0: warning: "enable_Z" redefined
 #define  enable_Z() do{ Z_enable();  Z2_enable();  Z3_enable();  Z4_enable(); }while(0)

In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:142:0: note: this is the location of the previous definition
 #define  enable_Z() do{ Z_enable(); Z2_enable(); Z3_enable(); }while(0)

In file included from Marlin\src\module/stepper.h:47:0,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/indirection.h:702:0: warning: "disable_Z" redefined
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); Z4_disable(); CBI(axis_known_position, Z_AXIS); }while(0)
 
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:143:0: note: this is the location of the previous definition
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); CBI(axis_known_position, Z_AXIS); }while(0)

In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:47:61: error: default argument given for parameter 1 of 'void manage_inactivity(bool)' [-fpermissive]
 void manage_inactivity(const bool ignore_stepper_queue=false);
                                                             ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:47:6: note: previous specification in 'void manage_inactivity(bool)' here
 void manage_inactivity(const bool ignore_stepper_queue=false);
      ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 1 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 2 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 3 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:69:43: error: default argument given for parameter 1 of 'void minkill(bool)' [-fpermissive]
 void minkill(const bool steppers_off=false);
                                           ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:326:6: note: previous specification in 'void minkill(bool)' here
 void minkill(const bool steppers_off=false);
      ^~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RC_bigtree_512K\src\src\Marlin.cpp.o] Error 1
```

qwewer0, 2020-01-27:
PlatformIO in MarlinCore.h underscores the "false" in these:

(47) - void manage_inactivity(const bool ignore_stepper_queue=false);

(68) - void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);

(69) - void minkill(const bool steppers_off=false);

qwewer0, 2020-01-27:
Unable to compile Bugfix-2.0.x with this terminal output:

```
Marlin\src\gcode\gcode.cpp: In static member function 'static void GcodeSuite::process_parsed_command(bool)':
Marlin\src\gcode\gcode.cpp:335:16: error: 'G60' was not declared in this scope
       case 60: G60(); break;                                      // G60:  save current position
                ^~~
Marlin\src\gcode\gcode.cpp:335:16: note: suggested alternative: 'G10'
       case 60: G60(); break;                                      // G60:  save current position
                ^~~
                G10
Marlin\src\gcode\gcode.cpp:336:16: error: 'G61' was not declared in this scope
       case 61: G61(); break;                                      // G61:  Apply/restore saved coordinates.
                ^~~
Marlin\src\gcode\gcode.cpp:336:16: note: suggested alternative: 'G11'
       case 61: G61(); break;                                      // G61:  Apply/restore saved coordinates.
                ^~~
                G11
*** [.pio\build\STM32F103RC_bigtree_512K\src\src\gcode\gcode.cpp.o] Error 1
```

tpruvot, 2020-01-27:
fix in #16692

qwewer0, 2020-01-27:
By commenting out lines 335 and 336 in gcode.cpp I was able to compile Bugfix-2.0.x. The problem with 2.0.2 is still remain.

Talha909, 2020-01-27:
Can't compile Marlin 2.0.2

Compiling .pio\build\LPC1769\src\src\core\utility.cpp.o
In file included from Marlin\src\module/stepper.h:47,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/indirection.h:701: warning: "enable_Z" redefined
 #define  enable_Z() do{ Z_enable();  Z2_enable();  Z3_enable();  Z4_enable(); }while(0)

In file included from Marlin\src\Marlin.cpp:31:
Marlin\src\Marlin.h:142: note: this is the location of the previous definition
 #define  enable_Z() do{ Z_enable(); Z2_enable(); Z3_enable(); }while(0)

In file included from Marlin\src\module/stepper.h:47,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/indirection.h:702: warning: "disable_Z" redefined
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); Z4_disable(); CBI(axis_known_position, Z_AXIS); }while(0)

In file included from Marlin\src\Marlin.cpp:31:
Marlin\src\Marlin.h:143: note: this is the location of the previous definition
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); CBI(axis_known_position, Z_AXIS); }while(0)
 
Compiling .pio\build\LPC1769\src\src\feature\I2CPositionEncoder.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\Max7219_Debug_LEDs.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\babystep.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\backlash.cpp.o
In file included from Marlin\src\lcd/ultralcd.h:40,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:47:61: error: default argument given for parameter 1 of 'void manage_inactivity(bool)' [-fpermissive]
 void manage_inactivity(const bool ignore_stepper_queue=false);
                                                             ^
In file included from Marlin\src\Marlin.cpp:31:
Marlin\src\Marlin.h:47:6: note: previous specification in 'void manage_inactivity(bool)' here
 void manage_inactivity(const bool ignore_stepper_queue=false);
      ^~~~~~~~~~~~~~~~~
Compiling .pio\build\LPC1769\src\src\feature\baricuda.cpp.o
In file included from Marlin\src\lcd/ultralcd.h:40,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 1 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 2 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 3 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:69:43: error: default argument given for parameter 1 of 'void minkill(bool)' [-fpermissive]
 void minkill(const bool steppers_off=false);
                                           ^
In file included from Marlin\src\Marlin.cpp:31:
Marlin\src\Marlin.h:326:6: note: previous specification in 'void minkill(bool)' here
 void minkill(const bool steppers_off=false);
      ^~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LPC1769\src\src\Marlin.cpp.o] Error 1
============================================================ [FAILED] Took 13.33 seconds ============================================================

Environment                   Status    Duration
----------------------------  --------  ------------
megaatmega2560                IGNORED
megaatmega1280                IGNORED
rambo                         IGNORED
FYSETC_F6_13                  IGNORED
FYSETC_F6_14                  IGNORED
sanguino_atmega644p           IGNORED
sanguino_atmega1284p          IGNORED
melzi                         IGNORED
melzi_optiboot                IGNORED
at90usb1286_cdc               IGNORED
at90usb1286_dfu               IGNORED
DUE                           IGNORED
DUE_USB                       IGNORED
DUE_debug                     IGNORED
LPC1768                       IGNORED
LPC1769                       FAILED    00:00:13.333

ellensp, 2020-01-27:
even with zero changes it fails to compile..
Issue is it has marlin.h, marlin.cpp and marlincore.h, marlincore.cpp...  cant have both sets 
Then you get the G60/G61 errors also...

Calimerorulez, 2020-01-27:
Same here:

```
Compiling .pio/build/STM32F103RC_bigtree/src/src/core/utility.cpp.o
In file included from Marlin/src/module/stepper.h:47:0,
                 from Marlin/src/Marlin.cpp:37:
Marlin/src/module/stepper/indirection.h:701:0: warning: "enable_Z" redefined
 #define  enable_Z() do{ Z_enable();  Z2_enable();  Z3_enable();  Z4_enable(); }while(0)
 
In file included from Marlin/src/Marlin.cpp:31:0:
Marlin/src/Marlin.h:142:0: note: this is the location of the previous definition
 #define  enable_Z() do{ Z_enable(); Z2_enable(); Z3_enable(); }while(0)
 
In file included from Marlin/src/module/stepper.h:47:0,
                 from Marlin/src/Marlin.cpp:37:
Marlin/src/module/stepper/indirection.h:702:0: warning: "disable_Z" redefined
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); Z4_disable(); CBI(axis_known_position, Z_AXIS); }while(0)
 
In file included from Marlin/src/Marlin.cpp:31:0:
Marlin/src/Marlin.h:143:0: note: this is the location of the previous definition
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); CBI(axis_known_position, Z_AXIS); }while(0)
 
In file included from Marlin/src/lcd/ultralcd.h:40:0,
                 from Marlin/src/Marlin.cpp:34:
Marlin/src/lcd/../MarlinCore.h:45:1: error: default argument given for parameter 1 of 'void idle(bool)' [-fpermissive]
 );
 ^
In file included from Marlin/src/Marlin.cpp:31:0:
Marlin/src/Marlin.h:41:6: note: previous specification in 'void idle(bool)' here
 void idle(
      ^~~~
In file included from Marlin/src/lcd/ultralcd.h:40:0,
                 from Marlin/src/Marlin.cpp:34:
Marlin/src/lcd/../MarlinCore.h:47:61: error: default argument given for parameter 1 of 'void manage_inactivity(bool)' [-fpermissive]
 void manage_inactivity(const bool ignore_stepper_queue=false);
                                                             ^
In file included from Marlin/src/Marlin.cpp:31:0:
Marlin/src/Marlin.h:47:6: note: previous specification in 'void manage_inactivity(bool)' here
 void manage_inactivity(const bool ignore_stepper_queue=false);
      ^~~~~~~~~~~~~~~~~
In file included from Marlin/src/lcd/ultralcd.h:40:0,
                 from Marlin/src/Marlin.cpp:34:
Marlin/src/lcd/../MarlinCore.h:68:106: error: default argument given for parameter 1 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin/src/Marlin.cpp:31:0:
Marlin/src/Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin/src/lcd/ultralcd.h:40:0,
                 from Marlin/src/Marlin.cpp:34:
Marlin/src/lcd/../MarlinCore.h:68:106: error: default argument given for parameter 2 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin/src/Marlin.cpp:31:0:
Marlin/src/Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin/src/lcd/ultralcd.h:40:0,
                 from Marlin/src/Marlin.cpp:34:
Marlin/src/lcd/../MarlinCore.h:68:106: error: default argument given for parameter 3 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin/src/Marlin.cpp:31:0:
Marlin/src/Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
compilation terminated due to -fmax-errors=5.
*** [.pio/build/STM32F103RC_bigtree/src/src/Marlin.cpp.o] Error 1
=============================================================== [FAILED] Took 190.07 seconds ===============================================================
[... 35 lines cut ...]
mks_robin                     IGNORED
mks_robin_pro                 IGNORED
mks_robin_lite                IGNORED
mks_robin_lite3               IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
STM32F401VE_STEVAL            IGNORED
FLYF407ZG                     IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_GTR_V1_0              IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
rumba32_f446ve                IGNORED
mks_rumba32                   IGNORED
include_tree                  IGNORED
========================================================== 1 failed, 0 succeeded in 00:03:10.074 ==========================================================
  #define STRING_DISTRIBUTION_DATE "2020-01-27"
```

DerAndere1, 2020-01-27:
Can you compile after deleting src/Marlin.h and src/Marlin.cpp files?

qwewer0, 2020-01-27:
> Can you compile after deleting files src/Marlin.h and src/Marlin.cpp ?

What files exactly?

Talha909, 2020-01-27:
Magic Works.  Whats the reason behind this . I deleted two files and after that compile the firmware. And its successful

DerAndere1, 2020-01-27:
If you use Marlin 2.0.2 release and cannot simply update to latest bugfix-2.0.x (wich was fixed with commit https://github.com/MarlinFirmware/Marlin/commit/e2eef1256a6feaceb8f4c7f02188e8d6d6d43519 ), I can confirm that it compiles if you delete files src/Marlin.h and src/Marlin.cpp and also delete lines 335 and 336 in file src/gcode/gcode.cpp

Hesi-Re, 2020-01-27:
Deleting lines 335 and 336 in file Marlin/src/gcode/gcode.cpp and also deleting files Marlin/src/Marlin.h and Marlin/src/Marlin.cpp doesn't solve the problem. I cannot compile with VSCode:

Marlin/src/_Marlin.cpp:31:10: fatal error: Marlin.h: No such file or directory

Lord-Quake, 2020-01-27:
Did you really delete the files?

spooky79, 2020-01-27:
I also have a compilation problem  , skr e3 mini
[image]

Scope666, 2020-01-27:
Same error as @spooky79     (SKR Mini E3 v1.2)

```
In file included from Marlin\src\module/stepper.h:47:0,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/indirection.h:701:0: warning: "enable_Z" redefined
Compiling .pio\build\STM32F103RC_bigtree_512K_USB\src\src\feature\backlash.cpp.o
 #define  enable_Z() do{ Z_enable();  Z2_enable();  Z3_enable();  Z4_enable(); }while(0)
 
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:142:0: note: this is the location of the previous definition
 #define  enable_Z() do{ Z_enable(); Z2_enable(); Z3_enable(); }while(0)
 
In file included from Marlin\src\module/stepper.h:47:0,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/indirection.h:702:0: warning: "disable_Z" redefined
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); Z4_disable(); CBI(axis_known_position, Z_AXIS); }while(0)

In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:143:0: note: this is the location of the previous definition
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); CBI(axis_known_position, Z_AXIS); }while(0)

In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:45:1: error: default argument given for parameter 1 of 'void idle(bool)' [-fpermissive]
 );
 ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:41:6: note: previous specification in 'void idle(bool)' here
 void idle(
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:47:61: error: default argument given for parameter 1 of 'void manage_inactivity(bool)' [-fpermissive]
 void manage_inactivity(const bool ignore_stepper_queue=false);
                                                             ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:47:6: note: previous specification in 'void manage_inactivity(bool)' here
 void manage_inactivity(const bool ignore_stepper_queue=false);
      ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 1 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 2 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 3 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RC_bigtree_512K_USB\src\src\Marlin.cpp.o] Error 1
```

Hesi-Re, 2020-01-27:
> Did you really delete the files?

I confirm deleting those files solves the problem. Before I just changed files name.

Hesi-Re, 2020-01-27:
I also get this compile error:

Marlin/src/feature/pause.cpp: In function 'bool unload_filament(const float&, bool, PauseMode)':
Marlin/src/feature/pause.cpp:355:20: error: 'FILAMENT_UNLOAD_PURGE_FEEDRATE' was not declared in this scope
                   (FILAMENT_UNLOAD_PURGE_FEEDRATE) * mix_multiplier);
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/feature/pause.cpp:355:20: note: suggested alternative: 'FILAMENT_UNLOAD_PURGE_RETRACT'
                   (FILAMENT_UNLOAD_PURGE_FEEDRATE) * mix_multiplier);
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                    FILAMENT_UNLOAD_PURGE_RETRACT

Scope666, 2020-01-27:
> I get also this compile error:
> 
> Marlin/src/feature/pause.cpp: In function 'bool unload_filament(const float&, bool, PauseMode)':
> Marlin/src/feature/pause.cpp:355:20: error: 'FILAMENT_UNLOAD_PURGE_FEEDRATE' was not declared in this scope
> (FILAMENT_UNLOAD_PURGE_FEEDRATE) * mix_multiplier);
> ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> Marlin/src/feature/pause.cpp:355:20: note: suggested alternative: 'FILAMENT_UNLOAD_PURGE_RETRACT'
> (FILAMENT_UNLOAD_PURGE_FEEDRATE) * mix_multiplier);
> ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> FILAMENT_UNLOAD_PURGE_RETRACT

I've been having to add this line in for awhile now, not sure why it's missing:

Lord-Quake, 2020-01-27:
You need to use the configs that are found in the Marlin folder and edit these to your needs. Don't use the ones found in the config/example folder. They don't seem to be updated for 2.0.2

Hesi-Re, 2020-01-27:
I just added this line to my configuration_adv.h file:

#define FILAMENT_UNLOAD_PURGE_FEEDRATE      10  // (mm/s) Unload purge filament feedrate. This can be pretty fast.

Now it compiles without errors!

spooky79, 2020-01-27:
> 
> 
> You need to use the configs that are found in the Marlin folder and edit these to your needs. Don't use the ones found in the config/example folder. They don't seem to be updated for 2.0.2

manually configured the same problem

DerAndere1, 2020-01-27:
If I modify Marlin 2.0.2, according to comment https://github.com/MarlinFirmware/Marlin/issues/16691#issuecomment-578858359 and replace the configs with an example config from https://github.com/MarlinFirmware/Configurations, I can compile sucessfully using the Arduino IDE.

Scope666, 2020-01-27:
> If I modify Marlin 2.0.2, according to comment [#16691 (comment)](https://github.com/MarlinFirmware/Marlin/issues/16691#issuecomment-578858359) and replace the configs with an example config from https://github.com/MarlinFirmware/Configurations, I can compile sucessfully using the Arduino IDE.

I thought we were encouraged to use VSCode / PlatformIO ?

DerAndere1, 2020-01-27:
I expect a working (re-)release, soon. If you do the same as I did (no additional modifications) and then use PlatformIO, does it still show the error that Marlin.h is missing? What does PlatformIO say if you keep Marlin.h and Marlin.cpp intact and instead delete MarlinCore.h and MarlinCore.cpp?

Lord-Quake, 2020-01-28:
Update:
- The G60/61 bug in the OP has been patched.

Download Marlin 2.0.2 again.

There is still the issue with Marlin.h and Marlin.cpp.
To solve this you must **delete** the files
`.../Marlin/src/Marlin.h`
 and
`.../Marlin/src/Marlin.cpp ` 
Do not rename nor copy them in a newly created folder within Marlin.

You should now be able to compile successfully. In my case with "STM32F103RC_bigtree_512K"
If you still have an issue then it's best you post your configs:
`Configuration.h`
`Configuration_adv.h`
`platformio.ini`

qwewer0, 2020-01-28:
20.01.28. Still can't compile 2.0.2. Will wait for official fix.

Lord-Quake, 2020-01-28:
Post your config files so we can take a look.

qwewer0, 2020-01-28:
See it on the first comment.

Lord-Quake, 2020-01-28:
I was able to successfully compile with your configs:

**Linking .pio\build\STM32F103RC_bigtree_512K\firmware.elf
Checking size .pio\build\STM32F103RC_bigtree_512K\firmware.elf
Building .pio\build\STM32F103RC_bigtree_512K\firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [====      ]  41.5% (used 20408 bytes from 49152 bytes)
PROGRAM: [======    ]  55.0% (used 288588 bytes from 524288 bytes)**

qwewer0, 2020-01-28:
> I was able to successfully compile with your configs:

And you didn't change anything?
I still get these terminal outputs:

```
In file included from Marlin\src\module/stepper.h:47:0,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/indirection.h:701:0: warning: "enable_Z" redefined
 #define  enable_Z() do{ Z_enable();  Z2_enable();  Z3_enable();  Z4_enable(); }while(0)
 
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:142:0: note: this is the location of the previous definition
 #define  enable_Z() do{ Z_enable(); Z2_enable(); Z3_enable(); }while(0)
 
In file included from Marlin\src\module/stepper.h:47:0,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/indirection.h:702:0: warning: "disable_Z" redefined
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); Z4_disable(); CBI(axis_known_position, Z_AXIS); }while(0)
 
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:143:0: note: this is the location of the previous definition
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); CBI(axis_known_position, Z_AXIS); }while(0)
 
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:47:61: error: default argument given for parameter 1 of 'void manage_inactivity(bool)' [-fpermissive]
 void manage_inactivity(const bool ignore_stepper_queue=false);
                                                             ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:47:6: note: previous specification in 'void manage_inactivity(bool)' here
 void manage_inactivity(const bool ignore_stepper_queue=false);
      ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 1 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 2 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 3 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:69:43: error: default argument given for parameter 1 of 'void minkill(bool)' [-fpermissive]
 void minkill(const bool steppers_off=false);
                                           ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:326:6: note: previous specification in 'void minkill(bool)' here
 void minkill(const bool steppers_off=false);
      ^~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RC_bigtree_512K\src\src\Marlin.cpp.o] Error 1
==================================================================================== [FAILED] Took 21.48 seconds
```

Lord-Quake, 2020-01-28:
Are you absolutely sure you did this:

There is still the issue with Marlin.h and Marlin.cpp.
To solve this you must delete the files
.../Marlin/src/Marlin.h
and
.../Marlin/src/Marlin.cpp
Do not rename nor copy them in a newly created folder within Marlin.

qwewer0, 2020-01-28:
I'm not sure about deleting files, I rather wait for a release with the two file deleted.

Rhinohorn58, 2020-01-28:
> 
> 
> 20.01.28. Still can't compile 2.0.2. Will wait for official fix.

Same,  can't compile
[Config+ ini.zip](https://github.com/MarlinFirmware/Marlin/files/4122080/Config%2B.ini.zip)

dok-net, 2020-01-28:
@thinkyhead The merge in ae2a0036eb97d08f65e7f1f5a6de1e6988ffddb9 has resurrected the Marlin.h and Marlin.cpp files that were renamed to MarlinCore.* in d0e1166cce60381fbdad1105d13765806a525d99.
To make things works, both the original and the renamed files have digressed from common base in both branches.
I would seem release 2.0.2 is broken?

Lord-Quake, 2020-01-28:
> > 20.01.28. Still can't compile 2.0.2. Will wait for official fix.
> 
> Same, can't compile
> [Config+ ini.zip](https://github.com/MarlinFirmware/Marlin/files/4122080/Config%2B.ini.zip)

You have a general config issue e.g.:
In file included from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/SanityCheck.h:1442:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."

Rhinohorn58, 2020-01-28:
> You have a general config issue e.g.:
In file included from Marlin\src\HAL\HAL_LPC1768../../core/../inc/MarlinConfig.h:38:0,
from Marlin\src\HAL\HAL_LPC1768../../core/serial.h:24,
from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
Marlin\src\HAL\HAL_LPC1768../../core/../inc/SanityCheck.h:1442:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
#error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."

Apologies, i changed my pinout to use the "EXTRUDER_AUTO_FAN_TEMPERATURE" feature.
pinout files attached

[lpc1768.zip](https://github.com/MarlinFirmware/Marlin/files/4122188/lpc1768.zip)

Thijsvanzwam, 2020-01-28:
Cannot compile either:

In file included from sketch/src/lcd/ultralcd.h:40:0,
                 from sketch/src/Marlin.cpp:34:
sketch/src/lcd/../MarlinCore.h: In function 'bool IsRunning()':
sketch/src/lcd/../MarlinCore.h:74:13: error: redefinition of 'bool IsRunning()'
 inline bool IsRunning() { return  Running; }
             ^
In file included from sketch/src/Marlin.cpp:31:0:
sketch/src/Marlin.h:331:13: note: 'bool IsRunning()' previously defined here
 inline bool IsRunning() { return  Running; }
             ^
In file included from sketch/src/lcd/ultralcd.h:40:0,
                 from sketch/src/Marlin.cpp:34:
sketch/src/lcd/../MarlinCore.h: In function 'bool IsStopped()':
sketch/src/lcd/../MarlinCore.h:75:13: error: redefinition of 'bool IsStopped()'
 inline bool IsStopped() { return !Running; }
             ^
In file included from sketch/src/Marlin.cpp:31:0:
sketch/src/Marlin.h:332:13: note: 'bool IsStopped()' previously defined here
 inline bool IsStopped() { return !Running; }
             ^
exit status 1
Fout bij het compileren voor board Anet V1.0 (Optiboot)

[config files.zip](https://github.com/MarlinFirmware/Marlin/files/4122286/config.files.zip)

dok-net, 2020-01-28:
@DerAndere1 
> If you use Marlin 2.0.2 release and cannot simply update to latest bugfix-2.0.x (wich was fixed with commit e2eef12 ), I can confirm that 2.0.2 relase compiles if you delete files Marlin/src/Marlin.h and Marlin/src/Marlin.cpp and also delete lines 335 and 336 in file Marlin/src/gcode/gcode.cpp

Touching gcode.cpp is completely unnecessary, also, it would have to be lines 335 through 337 or you break the #if/#endif pair :-)
Please note that the configuration files have seen a major update, but that these versions are now in the new [MarlinFirmware/Configurations](https://github.com/MarlinFirmware/Configurations)  repository only.
Compiling release 2.0.2 (tag) with the printer-specific Configuration.h/Configuration_adv.h from there works - besides the great quality of the gcc suite which crashes sometimes.

Balzac40, 2020-01-28:
@dok-net Following your instructions, 2.0.2 well compiles (some warnings as often). Thanks!

boelle, 2020-01-28:
people.... 2.0.2 is allready old compared to bugfix 2.0.x, try with that instead

Lord-Quake, 2020-01-28:
For those reluctant using nightly builds can safely use, as @boelle suggests, the bugfix 2.0.x. firmware version. In our case all the above mentioned issues have been taken care off.

dkeekstra, 2020-01-28:
What worked for me was deleting the Marlin.cpp, Marlin.h in the src folder and deleting line 335 and 336 in \src\gcode\gcode.cpp

boelle, 2020-01-28:
you dont have to delete anything

> deleting the Marlin.cpp, Marlin.h in the src folder and deleting line 335 and 336 in \src\gcode\gcode.cpp

dkeekstra, 2020-01-28:
> 
> 
> you dont have to delete anything
> 
> > deleting the Marlin.cpp, Marlin.h in the src folder and deleting line 335 and 336 in \src\gcode\gcode.cpp

If i dont delete Marlin.cpp and Marlin.h i got the 'marlincore.h:74:13: error: redefinition of 'bool IsRunning()'' error when compiling for AVR

boelle, 2020-01-28:
and you are of course using bugfix 2.0.x as suggested right?

qwewer0, 2020-01-28:
If this would be a problem in Bugfix-2.0.x then I would have write that in the title. If we get a new release or a patched 2.0.2, then this can be closed.

boelle, 2020-01-28:
2.0.2 will not be patched, it is fixed allready in 2.0.x
2.0.x will at some point be frozen and saved as 2.0.3

look at what @Lord-Quake wrote

qwewer0, 2020-01-28:
The problem is still present in 2.0.x

```
In file included from Marlin\src\module/stepper.h:47:0,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/indirection.h:701:0: warning: "enable_Z" redefined
 #define  enable_Z() do{ Z_enable();  Z2_enable();  Z3_enable();  Z4_enable(); }while(0)

In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:142:0: note: this is the location of the previous definition
 #define  enable_Z() do{ Z_enable(); Z2_enable(); Z3_enable(); }while(0)
 
In file included from Marlin\src\module/stepper.h:47:0,
                 from Marlin\src\Marlin.cpp:37:
Compiling .pio\build\STM32F103RC_bigtree_512K\src\src\feature\babystep.cpp.o
Marlin\src\module/stepper/indirection.h:702:0: warning: "disable_Z" redefined
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); Z4_disable(); CBI(axis_known_position, Z_AXIS); }while(0)
Compiling .pio\build\STM32F103RC_bigtree_512K\src\src\feature\backlash.cpp.o
 
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:143:0: note: this is the location of the previous definition
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); CBI(axis_known_position, Z_AXIS); }while(0)
 
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:47:61: error: default argument given for parameter 1 of 'void manage_inactivity(bool)' [-fpermissive]
 void manage_inactivity(const bool ignore_stepper_queue=false);
                                                             ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:47:6: note: previous specification in 'void manage_inactivity(bool)' here
 void manage_inactivity(const bool ignore_stepper_queue=false);
      ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 1 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 2 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 3 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:69:43: error: default argument given for parameter 1 of 'void minkill(bool)' [-fpermissive]
 void minkill(const bool steppers_off=false);
                                           ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:326:6: note: previous specification in 'void minkill(bool)' here
 void minkill(const bool steppers_off=false);
      ^~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RC_bigtree_512K\src\src\Marlin.cpp.o] Error 1
==================================================================================== [FAILED] Took 21.49 seconds
```

Lord-Quake, 2020-01-28:
@qwewer0 I think you need to reset.
2.0.x is a freezed release. It will not be changed. If you want the latest working release you will need to use Bugfix-2.0.x that has all the issues mentioned in this thread fixed.
So.... download it, use the configs found here https://github.com/MarlinFirmware/Configurations ,  Change them accordingly, install the FW on your printer and enjoy.

qwewer0, 2020-01-28:
Was able to compile Bugfix-2.0.x 20.01.28.
Now I just need to wait for 2.0.3.

boelle, 2020-01-28:
why? that will be a frozen copy as soon its out and 2.0.x will move on

but your call

will close this one

closed 2020-01-28 by boelle

kreuzhofer, 2020-01-28:
I just cloned a Bugfix-2.0.x into a new directory. Problem still exists. Even with a clean before build nothing changes. I would not confirm this is fixed in the 2.0.x.

kreuzhofer, 2020-01-28:
> Was able to compile Bugfix-2.0.x 20.01.28.
> Now I just need to wait for 2.0.3.

What did you do to make it compile. I cloned a fresh copy, used the Ender 3 configuration files and it does not build.

rs-development, 2020-01-28:
Same here, fresh copy does not compile with PlatformIO. This should be fixed!

qwewer0, 2020-01-28:
I usually don't update to Bugfix because of the potential bugs in that, that's why I waited for 2.0.2, but it still got one in. 

> why? that will be a frozen copy as soon its out and 2.0.x will move on
> 
> but your call
> 
> will close this one

I just cloned the latest Bugfix and my configs, it just worked. Don't know why wouldn't work for you.

> What did you do to make it compile. I cloned a fresh copy, used the Ender 3 configuration files and it does not build.
> 
> Even with default config, not using Ender 3 configs does not compile

qwewer0, 2020-01-28:
If this is still a problem, then we need to reopen it.

> will close this one

rs-development, 2020-01-28:
Should it compile with the default config files included in 2.0.x? If not, can you provide yours for testing?

qwewer0, 2020-01-28:
Try Bugfix-2.0.x, my configs are in the first comment.

> Should it compile with the default config files included in 2.0.x? If not, can you provide yours for testing?

rs-development, 2020-01-28:
> Try Bugfix-2.0.x, my configs are in the first comment.
> 
> > Should it compile with the default config files included in 2.0.x? If not, can you provide yours for testing?

It does compile fine! I assumed 2.0.x is ahead of 2.0.2 but it is not. All good but maybe make sure future releases can be at least built. That is the point of a release in my opinion.

dshokouhi, 2020-01-28:
I also just did a fresh build from bugfix-2.0.x and I get the exact same errors as noted here: https://github.com/MarlinFirmware/Marlin/issues/16691#issuecomment-579383660

[18 more replies]