MarlinFirmware/Marlin #16694: [BUG] Unable to compile

GurenWolf, 2020-01-27:
Hello everyone.

I am using a Bigtreetech SKR 1.3 board with the latest marlin firmware as of today and an Anycubic Kossel Plus with TMC2208 drivers.

Software used to compile is Visual Studio Code with PIO extension.

### Bug Description

Cannot compile Malrin firmware downloaded 27/1/2020 at 10:30.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4117974/Marlin.zip)

### Steps to Reproduce

Compile the firmware after custom configurations are done.

**Expected behavior:**

Successful compilation.

**Actual behavior:**

Error appears during compilation:

`Marlin\src\gcode\gcode.cpp: In static member function 'static void GcodeSuite::process_parsed_command(bool)':
Marlin\src\gcode\gcode.cpp:335:16: error: 'G60' was not declared in this scope
       case 60: G60(); break;                                      // G60:  save current position
                ^~~
Marlin\src\gcode\gcode.cpp:335:16: note: suggested alternative: 'G30'
       case 60: G60(); break;                                      // G60:  save current position
                ^~~
                G30
Marlin\src\gcode\gcode.cpp:336:16: error: 'G61' was not declared in this scope
       case 61: G61(); break;                                      // G61:  Apply/restore saved coordinates.
                ^~~
Marlin\src\gcode\gcode.cpp:336:16: note: suggested alternative: '._61'
       case 61: G61(); break;                                      // G61:  Apply/restore saved coordinates.
                ^~~
                ._61
Compiling .pio\build\LPC1768\src\src\gcode\geometry\G53-G59.cpp.o
*** [.pio\build\LPC1768\src\src\gcode\gcode.cpp.o] Error 1`

Just in case the report is not to clear in text, here is an image:

[image]

sjasonsmith, 2020-01-27:
That is impacting every build right now. Just delete those two lines from gcode.cpp.

spooky79, 2020-01-27:
I also have a compilation problem , skr e3 mini
[image]

Lord-Quake, 2020-01-27:
Read through this thread to solve the compiling problems: https://github.com/MarlinFirmware/Marlin/issues/16691

spooky79, 2020-01-27:
manually configured the same problem

pekezu, 2020-01-27:
Can you help me too?

python version 2.7.16
build_type:   build
starting platformio
Exception in thread Thread-1:
Traceback (most recent call last):
  File "c:\python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "c:\python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "buildroot/share/atom/auto_build.py", line 1021, in <lambda>
    self.secondary_thread = threading.Thread(target = lambda q, arg1: q.put(run_PIO(arg1)), args=(que, ''))
  File "buildroot/share/atom/auto_build.py", line 838, in run_PIO
    pio_subprocess = subprocess.Popen(['platformio', 'run', '-e', target_env], stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
  File "c:\python27\lib\subprocess.py", line 394, in __init__
    errread, errwrite)
  File "c:\python27\lib\subprocess.py", line 644, in _execute_child
    startupinfo)
WindowsError: [Error 2] Mõõritettyõ tiedostoa ei l÷ydy (="file not found")

spooky79, 2020-01-27:
In file included from Marlin\src\module/stepper.h:47:0,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/indirection.h:701:0: warning: "enable_Z" redefined
 #define  enable_Z() do{ Z_enable();  Z2_enable();  Z3_enable();  Z4_enable(); }while(0)

In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:142:0: note: this is the location of the previous definition
 #define  enable_Z() do{ Z_enable(); Z2_enable(); Z3_enable(); }while(0)

In file included from Marlin\src\module/stepper.h:47:0,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/indirection.h:702:0: warning: "disable_Z" redefined
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); Z4_disable(); CBI(axis_known_position, Z_AXIS); }while(0)

In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:143:0: note: this is the location of the previous definition
 #define disable_Z() do{ Z_disable(); Z2_disable(); Z3_disable(); CBI(axis_known_position, Z_AXIS); }while(0)

Compiling .pio\build\STM32F103RC_bigtree_512K\src\src\feature\bedlevel\abl\abl.cpp.o
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:45:1: error: default argument given for parameter 1 of 'void idle(bool)' [-fpermissive]
 );
 ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:41:6: note: previous specification in 'void idle(bool)' here
 void idle(
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:47:61: error: default argument given for parameter 1 of 'void manage_inactivity(bool)' [-fpermissive]
 void manage_inactivity(const bool ignore_stepper_queue=false);
                                                             ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:47:6: note: previous specification in 'void manage_inactivity(bool)' here
 void manage_inactivity(const bool ignore_stepper_queue=false);
      ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 1 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 2 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h:68:106: error: default argument given for parameter 3 of 'void kill(const char*, const char*, bool)' [-fpermissive]
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
                                                                                                          ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:325:6: note: previous specification in 'void kill(const char*, const char*, bool)' here
 void kill(PGM_P const lcd_error=nullptr, PGM_P const lcd_component=nullptr, const bool steppers_off=false);
      ^~~~
compilation terminated due to -fmax-errors=5.

GurenWolf, 2020-01-27:
I agree with you @sjasonsmith, it's just one after another. I will follow the advice and see what happens.

GadgetAngel, 2020-01-28:
I downloaded the Marlin-2.0 RELEASE Version and it is getting ERRORS when compiling.  No edits made just a straight compile.  
[image]

ellensp, 2020-01-28:
DELETE Marlin.h and Marlin.cpp if Marlincore.h exists.  Don't just rename it.

thinkyhead, 2020-01-28:
The bug in the OP is now patched.

The other issue is due to the presence of obsolete files. If you're using a fork of Marlin that you have open in the GitHub Desktop app you can in the left column where unwanted changes have been made, including old files falling in by mistake, and you can use the "Discard Changes" command to undo them. If not using GitHub Desktop then follow your usual file management methods.

closed 2020-01-28 by thinkyhead