MarlinFirmware/Marlin #16695: [FR] Marlin Torch Height Control for CNC Plasma

KaleDobrohoczki, 2020-01-27:
### Description

Torch height control is a feature that allows a CNC plasma cutter to make Z height adjustments on the fly. The adjustments are made based on a voltage(isolated and safe) sent to the mainboard. Some plasma cutters have voltage dividers that can output small amounts of current. I have a simple DIY voltage divider capable of outputting between 0-5v. the output can be adjusted. The further the plasma cutter moves away from the material the higher the voltage gets. Plasma cutters work optimally at a specific height above the sheet metal. Heat and stresses in the material cause warping as cuts are being made. Movements are made between cuts where the torch is not cutting. These movements are made at a higher Z than the cuts. The adjustments don't need to be made while during these travels. I have a file that works on my MPCNC Lowrider2 plasma but is very challenging due to heat build-up and warpage.
 [SnowFlakePlasma.txt](https://github.com/MarlinFirmware/Marlin/files/4118623/SnowFlakePlasma.txt)
I have been able to cut successfully but edge quality and ease of use suffer from the lack of THC. I have also scrapped a lot of steel due to failures. 

### Feature Workflow

1. The user must choose a setpoint voltage that corresponds to the optimal cutting height. This could be done through the LCD screen, a Gcode command or both. This value may vary based on material type, thickness and the power level of the plasma cutter. 

2. As the CNC cuts, the voltage will vary as the material warps. If the voltage increases this means the torch is far away from the material and Z must be moved down. And vice versa. 

3. The z movements should not affect the homing. The movements could be made similar to babysteps on a 3d printer.

4. No adjustments should be made while the torch is not cutting. 

#### Additional Information
A similar discussion was made here:
https://github.com/MarlinFirmware/Marlin/issues/7562

@Roxy-3D expressed interest in this kind of feature. I spoke with @Duinishuodedui on another forum and he explained that he couldn't figure out how to build the voltage divider necessary for testing. So the project never really got going. But if @Roxy-3D is still out there an interested I have a voltage divider and a plasma CNC running Marlin to test things out.  

I think it might be possible to simulate the voltage of a plasma cutter by using the voltage of the fan pins on a ramps board. You could connect wires from the fan into whatever port is set to receive the voltage (I am clueless about how to actually code or configure that). You could either send Gcode with varying fan speeds or adjust the speeds through the LCD screen. This way just about anyone with a 3d printer can test the code. You could see if the z height can be varied based on a given voltage.

GMagician, 2020-01-27:
Don't know if hobbistic plasma works like the industrial one, latter has a well defined action.
1) usually move till it touch the sheet and set Z=0
2) raises to a percing height
3) ignit plasma
4) wait piercing time
5) move to cutting height
6) move XY axis
7) after some space it sample arc voltage and then try to keep it at same value during all cut.
8) Just before it ends cut it stop this "arc following" because of missing material that will fake arc reading

THC (Torch height control) may be done using babystepping I think but don't know how to really realize it

KaleDobrohoczki, 2020-01-28:
@GMagician, 

My Machine operats quite similarly:  
> 1. usually move till it touch the sheet and set Z=0 (Marlin can do this with either a probe or end stop. I haven't hooked that kind of thing up yet)
> 2. raises to a piercing height (my code has a definate height but without probing the CNC can't be exactly where it should be)
> 3. ignite plasma (yep)
> 4. wait piercing time (this is defined by the Gcode)
> 5. move to cutting height (set by Gcode but THC is needed to stay true)
> 6. move XY axis (Yep)
> 7. after some space it sample arc voltage and then try to keep it at same value during all cut. (this is perhaps a better way of finding a voltage set point than having a user input the value. I know not all commercial tables use this method. Some are user-defined) 
> 8. Just before it ends cut it stop this "arc following" because of missing material that will fake arc reading ( I did not know this but I can see how this would prevent crashes)

MarkMan0, 2020-01-29:
Could be implemented as a simple P controller, where the desired speed would be proportional to the error between the set and actual voltage

KaleDobrohoczki, 2020-01-29:
> Could be implemented as a simple P controller, where the desired speed would be proportional to the error between the set and actual voltage

The head of the torch only needs to go up or down by a few millimetres or less along mostly gradual changes in height. So the speed could be constant and as long as it can read the voltage and give steps quickly enough to adapt to the warping material. I'm not positive it would matter if the speed changes. I don't know enough about how to code to know if this method is more or less complicated.

KaleDobrohoczki, 2020-01-29:
@MarkMan0 your PID controller seems relevant to the kind of thing I'm looking for. Just with steppers, not a soldering iron.

GMagician, 2020-01-30:
> Could be implemented as a simple P controller, where the desired speed would be proportional to the error between the set and actual voltage

Yes, industrial machines use a P on speed to move torch to proper height. Not all machines use blind area (if H is inside this area, no thc following error is disabled). This area is used to stabilize up/down movements. Usually industrial machines use also a "manual" arc adjustment. Something like "babystepping" is doing with FFM printers.
Only one doubt about using babystepping to follow arc is that due to material planarity motors may change movement direction...what if babystepping is already moving one direction and that direction has to change?

KaleDobrohoczki, 2020-01-30:
>what if babystepping is already moving one direction and that direction has to change?

Oh yes, like a sharp corner or an acute angle. These kind of cuts are challenging even on commercial machines until it is dialed in. 1.5mm is the recommended cutting height. It's fine if the torch is closer or further for a moment. As long as it doesn't crash into the material the cut won't be failed. And if the torch strays to high it won't cut. If the steppers are moving too fast or if the travel given is too large it could possibly crash. Straying away from the optimal 1.5mm makes more dross to grind but not a failure.
I don't know how fast a Ramps board can send and receive information so hopefully it can adapt fast enough. I have been cutting at about 1800mm/min.

GMagician, 2020-01-30:
> Oh yes, like a sharp corner or an acute angle. These kind of cuts are challenging even on commercial 

To solve this commercial machines stop THC when speed goes lower than a % of the feed rate.

KaleDobrohoczki, 2020-01-30:
> > Oh yes, like a sharp corner or an acute angle. These kind of cuts are challenging even on commercial
> 
> To solve this commercial machines stop THC when speed goes lower than a % of the feed rate.

Is that possible in Marlin? Are there and features that operate in a similar way already? that kind of condition would be far superior to external THCs that come with a lot of plasma CNC out there.

KaleDobrohoczki, 2020-02-01:
I have a decent understanding of the process and methods used for THC but I don't have the knowledge of Marlin to code it. So if anyone could point me toward some tutorials and resources to learn I would be very grateful. Examples or explanations would be helpful. 
I don't know how to assign pins to work as inputs.
I don't know how to move the steppers. (sample code would be helpful)
What value can the pin read? Is it from 0-255 digitally? Does 255 equal 12v?
What is the specific code to just spin a motor? I can do this with Gcode but I don't know how to otherwise.

closed 2020-06-30 by github-actions[bot]

thinkyhead, 2020-07-13:
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y