MarlinFirmware/Marlin #16698: [BUG] BIGTREETRECH SKR V1.3 32bit heatbed wont heat, also no sensor value

CKCMorgoth, 2020-01-28:
Hi all
I have a problem with my SKR V1.3 board and marlin, my heatbed wont heat up, also the sensor will only show 0°C. Connections and cables are correct. My hotend works really fine. 
When I switch the hotend temp sensor and heatbed sensor, it will display the actual bed temp as hotend temp, so sensor is working. Also the heatbed is brandnew, and my multimeter reveals the bed gets no power,

So I wanted to check the firmware, I am using the github marlin2.0 for this board.
Searching the entries "TEMP_BED_PIN" or "HEATER_BED_PIN" in the pins_BTT_SKR_V1_3.h resulted in no entries, and in the  pins.h i just found:
#ifndef HEATER_0_PIN
#define HEATER_0_PIN -1
#endif
#ifndef HEATER_BED_PIN
#define HEATER_BED_PIN -1
#endif
#ifndef TEMP_0_PIN
#define TEMP_0_PIN -1
#endif
#ifndef TEMP_BED_PIN
#define TEMP_BED_PIN -1
#endif

So there are no actual entries defining the heating pins. So I guess the pins are defined somewhere else. I am using visual studio code but I cannot find any entries for the pins.

Also I added these lines to the pins_BTT_SKR_V1_3.h:
#define HEATER_0_PIN P2_07
#define HEATER_BED_PIN P2_05
#define TEMP_0_PIN P0_24
#define TEMP_BED_PIN P0_23 
But it doesn't change anything.
Can I set the pin for the heatbed byhand to check the board, or read in the pin for the sensor?

Can someone help please?
Thanks

Kokorone, 2020-01-28:
Check the sensor area around line 410 on configuration.h

You should have a value for both your hotend sensor and your bed sensor. For example
#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_5 0
#define TEMP_SENSOR_6 0
#define TEMP_SENSOR_7 0
#define TEMP_SENSOR_BED 1
#define TEMP_SENSOR_PROBE 0
#define TEMP_SENSOR_CHAMBER 0

ellensp, 2020-01-28:
pins_BTT_SKR_V1_3.h includes the file pins_BTT_SKR.h
All the other defines are in there.
If the controller cannot read the bed temperature it will not even try and heat it. its a safety feature.
You need to set #define TEMP_SENSOR_BED to a number other than 0
You also need to follow instructions and include your configuration files!

CKCMorgoth, 2020-01-28:
Big thumbs up to ellensp and Kokorone
I did not know that the bed wont heat up if the sensor is incorrect
I changed  TEMP_SENSOR_BED and now its working
Also big thx for the hint with the pins_BTT_SKR.h file, also something new and good to know.
So this can be closed, great community :)

closed 2020-01-28 by CKCMorgoth

thinkyhead, 2020-01-28:
> I did not know that the bed wont heat up if the sensor is incorrect

If `TEMP_SENSOR_BED` is 0 it means "no bed."