MarlinFirmware/Marlin #16700: [BUG] wrong home position SKR Pro 1.1 TMC 2209

Lino77, 2020-01-28:
Following problem:
my probe is here:
#define NOZZLE_TO_PROBE_OFFSET { -20, -40, 0 }
my Z safe home ist here 
#define Z_SAFE_HOMING_X_POINT (25)    
#define Z_SAFE_HOMING_Y_POINT (33)  
my 0 point 
#define X_MIN_POS -8
#define Y_MIN_POS -10
#define HOMING_BACKOFF_MM { 8, 10, 2 }
when homing it runs on x 45 and y 73. it shouldn't actually on x 25 and y 33. apparently it recognizes the negative value in offset as positive.
my bed is a bit bigger than specified in the config, the probe is not outside the bed with the named coordinates
 
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/4120820/Config.zip)

ellensp, 2020-01-28:
NOZZLE_TO_PROBE_OFFSET { -20, -40, 0 }
So your probe is 20mm left and 40mm back from your extruder nozzle.
 
Z_SAFE_HOMING_X_POINT, Z_SAFE_HOMING_Y_POINT is the position for the probe

you requested x25, but for your probe to get to 25 your extruder nozzle needs to move to x45
you requested y33 but for your probe to get to 33 your extruder nozzle needs to move to y73

boelle, 2020-01-28:
this is most likely a config error so closing as its not a bug afterall

closed 2020-01-28 by boelle

Lino77, 2020-01-28:
Okay, that was your mistake by me, I assumed that z safe homing concerns the nozzle and not on the Probe.