MarlinFirmware/Marlin #16703: [BUG] MKS SGEN L + Fysetc Mini 12864 = no characters displayed

bartolomeus, 2020-01-28:
### Bug Description
With the combo MKS SGEN L and the Fysetc mini 12864 the following happens after defining the display and neopixel support and the startup test) in config.h:
- the screen lights up, goes through the different colors of the startup test
- no characters are displayed, no bootscreen, nothing
- the reset pin works
- sd card are visible in repetier

So, after many attemps I found that undefining //#define FORCE_SOFT_SPI in PINS_MKS_SGEN_L.h
works. It brings the characters to the display. Also both SD card readers work.

### My Configurations
Marlin 2.01
MKS-SGEN_L
Fysetc Mini 12864 V2.1

[mkssgenlfysetc.zip](https://github.com/MarlinFirmware/Marlin/files/4123369/mkssgenlfysetc.zip)

### Steps to Reproduce
1. in configuration.h: 
#define FYSETC_MINI_12864_2_1
#define NEOPIXEL_LED
#define NEOPIXEL_TYPE   NEO_RGB 
#define NEOPIXEL_PIN     P0_17 
2. in configuration_adv.h
#define LED_CONTROL_MENU
#define LED_USER_PRESET_STARTUP
3. in pins.MKS_SGEN_L.h:
//#define FORCE_SOFT_SPI

**Expected behavior:** [What you expect to happen]
Characters displayed on LCD

**Actual behavior:** [What actually happens]
No characters on LCD (solved by editing the pins file)

#### Additional Information
The 'release media' entry in the LCD menu is very slow to react. Not sure if this is normal or expected bahaviour.

boelle, 2020-02-08:
do you still get this error with latest bugfix 2.0.x ?

bartolomeus, 2020-02-08:
I haven't tried with the latest bugfix. I am still on 2.0.1.

boelle, 2020-02-09:
try it

bartolomeus, 2020-02-11:
Tried with todays bugfix. I compiles with 1 (different) error:
'u8g_i2c_opt' defined but not used [-Wunused-variable]
[bugfix20200211.zip](https://github.com/MarlinFirmware/Marlin/files/4187832/bugfix20200211.zip)

ManuelMcLure, 2020-02-11:
> Tried with todays bugfix. I compiles with 1 (different) error:
> 'u8g_i2c_opt' defined but not used [-Wunused-variable]
> [bugfix20200211.zip](https://github.com/MarlinFirmware/Marlin/files/4187832/bugfix20200211.zip)

That's not an error, just a warning. The compiled firmware should be fine to use.

bartolomeus, 2020-02-11:
Yes, it does compile. And the warning has disappeared again.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle