MarlinFirmware/Marlin #16704: G28 Reboots printer in commit e4eaf32

phoenixfire425, 2020-01-28:
### Description

### Steps to Reproduce

1. pulled clean e4eaf32 latest bug fix
2. created new configs based of examples
3. ran M502
4. flashed new firmware
5. ran G28 - Printer reboots

I am using SKR 1.3 with Tevo Tornado Configs
BL Touch
LV8729  XYZE
Linear Advance
S Curve
Junction Deveation

I pulled 0c3cae5 and this issue does not happen.

boelle, 2020-01-28:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

phoenixfire425, 2020-01-28:
Here is the requested config.
I thought it was attached before but it looks like it did not upload.  sorry about that!

[Marlin_2.0.2_configs.zip](https://github.com/MarlinFirmware/Marlin/files/4124717/Marlin_2.0.2_configs.zip)

ogsv, 2020-01-30:
I'm having the same issue on a BTT SKR PRO board. Sorry I'm not able to upload my configs, can't remember where I built it from.

boelle, 2020-01-31:
@phoenixfire425 i had a look again and that commit was 100% esp32 related so since use an SKR it should not effect you

has the updated in the last 2 days made any changes ?

boelle, 2020-02-02:
@phoenixfire425 i had a look again and that commit was 100% esp32 related so since use an SKR it should not effect you

has the updates in the last 5 days made any changes ?

boelle, 2020-02-08:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

ogsv, 2020-02-08:
Issue is gone as of 2.0.3

closed 2020-02-08 by boelle