MarlinFirmware/Marlin #16709: fails to compile Formbot T-Rex 3

MickyPf, 2020-01-28:
Marlin 2.0.2
I used the standard config from the directory Samples / Formbot / T-Rex_3 for compaling.  

Executing task: C:\Users\pfeif\.platformio\penv\Scripts\platformio.exe run <

Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR 2.0.0 > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
PACKAGES:
 - framework-arduino-avr 5.0.0
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 17 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.2
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.3.4
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit MAX31865 library> 1.0.3
|   |-- <SPI> 1.0
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <TMC26XStepper>
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio\build\megaatmega2560\src\src\HAL\HAL_AVR\HAL.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\HAL_AVR\HAL_SPI.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\HAL_AVR\MarlinSerial.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\HAL_AVR\Servo.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\HAL_AVR\fast_pwm.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\HAL_AVR\persistent_store_eeprom.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\HAL_AVR\u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\HAL_AVR\watchdog.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\shared\eeprom_i2c.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\shared\eeprom_spi.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\shared\persistent_store_api.cpp.o
Compiling .pio\build\megaatmega2560\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\megaatmega2560\src\src\Marlin.cpp.o
Compiling .pio\build\megaatmega2560\src\src\MarlinCore.cpp.o
Compiling .pio\build\megaatmega2560\src\src\core\serial.cpp.o
Compiling .pio\build\megaatmega2560\src\src\core\utility.cpp.o
Compiling .pio\build\megaatmega2560\src\src\feature\I2CPositionEncoder.cpp.o
Compiling .pio\build\megaatmega2560\src\src\feature\Max7219_Debug_LEDs.cpp.o
Compiling .pio\build\megaatmega2560\src\src\feature\babystep.cpp.o
Compiling .pio\build\megaatmega2560\src\src\feature\backlash.cpp.o
Compiling .pio\build\megaatmega2560\src\src\feature\baricuda.cpp.o
Compiling .pio\build\megaatmega2560\src\src\feature\bedlevel\abl\abl.cpp.o
Compiling .pio\build\megaatmega2560\src\src\feature\bedlevel\bedlevel.cpp.o
[... 74 lines cut ...]
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:326:6: note: previous specification in 'void minkill(bool)' here
 void minkill(const bool steppers_off=false);
      ^
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h: In function 'bool IsRunning()':
Marlin\src\lcd/../MarlinCore.h:74:13: error: redefinition of 'bool IsRunning()'
 inline bool IsRunning() { return  Running; }
             ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:331:13: note: 'bool IsRunning()' previously defined here
 inline bool IsRunning() { return  Running; }
             ^
In file included from Marlin\src\lcd/ultralcd.h:40:0,
                 from Marlin\src\Marlin.cpp:34:
Marlin\src\lcd/../MarlinCore.h: In function 'bool IsStopped()':
Marlin\src\lcd/../MarlinCore.h:75:13: error: redefinition of 'bool IsStopped()'
 inline bool IsStopped() { return !Running; }
             ^
In file included from Marlin\src\Marlin.cpp:31:0:
Marlin\src\Marlin.h:332:13: note: 'bool IsStopped()' previously defined here
 inline bool IsStopped() { return !Running; }
             ^
*** [.pio\build\megaatmega2560\src\src\Marlin.cpp.o] Error 1

boelle, 2020-01-28:
not sure if this one is related but it might be
https://github.com/MarlinFirmware/Marlin/issues/16691

dok-net, 2020-01-29:
@boelle Definitely Marlin.cpp, Marlin.h and MarlinCore.cpp, MarlinCore.h are all in the build tree of the reporter.
Contrary to what you have been assuming, there MUST a patch done to the 2.0.x branch, of which tag 2.0.2, the current release, is a commit.
That's because merging, as is the case at release time, bugfix-2.0.x and 2.0.x at the time of writing this, again results in the duplication of Marlin.* and MarlinCore.* in Marlin/src. So please desist from closing PRs and issues to this effect and help us get this over with.
FYI: I am the person who requested the rename from Marlin.* to MarlinCore.* in the first place due to incompatibilities with VisualMicro Arduino IDE for Visual Studio :-)

InsanityAutomation, 2020-01-29:
I really should update this config a bit with what I'm using nowadays lol maybe when I'm done playing with G425 setup on it....

ellensp, 2020-02-01:
This was the Marlin.h and MarlinCore.h issue..  This has been resolved.
Grab the current Marlin and it should fix this issue for you.

closed 2020-02-01 by thinkyhead

dok-net, 2020-02-01:
@thinkyhead Could you please reconsider your handling of branches and tags? It's real mess for forks and closes right now, even for release 2.0.3 there's a commit on branch 2.0.x, message "Version 2.0.3", e6a7be922b300bc68030a5d301743ac30d6d1834, but there's also a tagged (2.0.3) commit without branch (!), 8eb6603e2aa948f54f1e6faec264f7309b214a89, bother inevitably differ...
Basically, please tag on branch 2.0.x, and never ever for such a wellknown project force-push or delete / change tags.