MarlinFirmware/Marlin #16710: Enabling SD support and/or EEPROM has made SKR 1.4 Turbo unresponsive

crm552, 2020-01-28:
### Description

After enabling SD support and EEPROM support in Marlin, compiling, and flashing to SKR 1.4 Turbo, display is powered up, but blank.  I've checked cables, connectors, verified the display is working on another board.  I've disconnected everything from the board except the display and still can't get anything to show up, even with a basic bare bones marlin configuration.  Board was working fine before enabling these 2 features, now it won't even show up via USB.

### Steps to Reproduce

1. Compile marlin for SKR 1.4 Turbo using CR-10 stock display.

2.  Flash marlin to SKR 1.4 Turbo

**Expected behavior:** EEPROM and SD support enabled along with working display

**Actual behavior:** No display, firmware.bin is still replaced with firmware.cur as expected, but no observable indication that firmware is running on the board.

boelle, 2020-01-28:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

crm552, 2020-01-29:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4125734/config.zip)

This is one of about 20 recompiles I did trying to get anything to work, disabiling features I already had working as I tried to make the configuration more and more simple.

boelle, 2020-01-29:
the maintainers made a boo-boo and bugfix 2.0.x does not compile, a lot of people have problems

i'm not a coder or maintainer so cant tell when it will work again

boelle, 2020-02-08:
@crm552 have you retested since your last comment?

PapiOrso, 2020-02-09:
I solved the problem by reloading the original firmware provided in the micro SD. I used a usb card reader to rename the original file from .CUR to .BIN then I inserted the SD in the board.

closed 2020-02-10 by thinkyhead