MarlinFirmware/Marlin #16711: [BUG] G76/Probe temperature calibration broken

thisiskeithb, 2020-01-29:
### Bug Description

When issuing `G76` for [Probe Temperature Calibration](https://marlinfw.org/docs/gcode/G076.html) on a Prusa MK3 with BigTreeTech's BTT002 board (with PINDA V2 probe), the printer homes and then I get the following serial output:

```log
Send: G76
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:104.00 Y:101.25 Z:5.00 E:0.00 Count X:10400 Y:10125 Z:2000
Recv: Waiting for printer to cool down.
```

The printer & bed are completely cold/this was issued after start up and waiting hours doesn't do anything.

### My Configurations

[Link to latest BTT002/MK3/TMC2209 config](https://github.com/thisiskeithb/Marlin/tree/73fc02348f63cfc51c92d62ff77b73ae0fb1da32) rebased on top of 9b45f34, but the relevant bits are below:

```c++
    // Max temperature that can be reached by heated bed.
    // This is required only for the calibration process.
    #define PTC_MAX_BED_TEMP 122 // BED_MAXTEMP

    // Park position to wait for probe cooldown
    #define PTC_PARK_POS_X 255.0F // X_MAX_POS
    #define PTC_PARK_POS_Y -4.0F  // Y_MIN_POS
    #define PTC_PARK_POS_Z 105.0F // (Z_MAX_POS/2)

    // Probe position to probe and wait for probe to reach target temperature
    #define PTC_PROBE_POS_X 127.5F // (X_BED_SIZE/2)
    #define PTC_PROBE_POS_Y 106.3F // (Y_BED_SIZE/2)
```

### Steps to Reproduce

1. [Meet requirements for `G76`/Probe temperature calibration](https://github.com/MarlinFirmware/Marlin/blob/9b45f342ee8ec1765cc564ffb811755c366cc50c/Marlin/src/gcode/calibrate/G76_M871.cpp#L40-L82)
2. Issue `G76`
3. Observe `Waiting for printer to cool down.` message via serial & wait forever as calibration never begins

**Expected behavior:** Calibration process to start

**Actual behavior:** Calibration process never starts

Tagging @tompe-proj as this is a beta feature and hopefully they are around to help troubleshoot.

TheNeskik, 2020-01-29:
Hello, 

I think there is a mistake in probe_temp_compensation.h (src/feature/)
You just need to change some value, exemple :

    static constexpr temp_calib_t cali_info_init[TSI_COUNT] = {
        {  10, 5, 30,  30 + 10 *  5 },       // Probe
        {  10, 5, 60,  60 + 10 *  5 },       // Bed
      #if ENABLED(USE_TEMP_EXT_COMPENSATION)
        { 180,  5, 20, 180 +  5 * 20 }        // Extruder
      #endif
    };
    static const temp_calib_t `cali_info[TSI_COUNT];

I didn't change value for the extruder. I've made those change in my config  :)

thisiskeithb, 2020-01-30:
Closing since the above PR fixed the reported problem and an unreported bug.

closed 2020-01-30 by thisiskeithb