MarlinFirmware/Marlin #16714: Cambio LCD anet A8 con lcd 12864 anet

bera99, 2020-01-29:
Ciao a tutti, dopo aver installato Marlin guardando questa guida 
https://youtu.be/wRODgnAqp1A e sbloccando la scheda con usbasp ho voluto cambiare schermo LCD dell'anet mettendo quello 12864 dell'anet A6 guardando questa guida https://youtu.be/uMsIb4Gwm3o ma dopo aver caricato lo sketch questo è come si vede sullo schermo nuovo, come risolvo?
[image]
[image]

boelle, 2020-01-29:
please use english

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

ellensp, 2020-01-29:
google translate failed to identify the Italian on this page...  had to force it.
In Configuration.h 

#define ST7920_DELAY_1 DELAY_NS(125)
#define ST7920_DELAY_2 DELAY_NS(125)
#define ST7920_DELAY_3 DELAY_NS(125)

boelle, 2020-01-29:
so this is a support q and not a bug?

ellensp, 2020-01-29:
Yes.

boelle, 2020-01-29:
closing it, should have been posted in discord

adress/join link is in the pinned post

closed 2020-01-29 by boelle