MarlinFirmware/Marlin #16719: [BUG] Extruder motor doens't work with marlin 2.0.2

bananafell3r, 2020-01-29:
Hi guys, 

I have an alfawise U10 (motherboard RAMPS PLUS), I installed marlin 2.0.2 but the extruder doesn't work. Do someone have this problem? how I can fix it?

reloxx13, 2020-01-29:
did u heat up the nozzle before moving e ?...

bananafell3r, 2020-01-29:
Yeah, obviously... And I tried, also the runout sensor doesn't work...

bananafell3r, 2020-01-29:
My board is exactly 3dymy ramps 1.4 plus

https://reprap.org/wiki/Ramps1.4plus

ellensp, 2020-01-30:
The page https://reprap.org/wiki/Ramps1.4plus has a circuit diagram that shows same pin out as ramps1.4. But the file pins_RAMPS_PLUS.h moves e0 pins (and others) elsewhere.
Since your e0 isnt working I wonder if there are more than one version of the ramp 1.4 plus...   try  MOTHERBOARD BOARD_RAMPS_14_EFB

bananafell3r, 2020-01-30:
Yes, the extruder works now.. But the filament sensor doesn't work.. I changed the pin sensor in 11, I am trying now to set the default pin 4.

ellensp, 2020-01-30:
Either way, it is not a bug in marlin, but just a configuration issue. please close this Bug report.

bananafell3r, 2020-01-30:
I solved...I close it

closed 2020-01-30 by bananafell3r