MarlinFirmware/Marlin #16721: [BUG] Fail to compile SKR PRO with autofan enabled in PWM mode

zalex94, 2020-01-29:
### Bug Description

Marlin 2.0 Bugfix Branch : Unable to compile SKR PRO 1.1 with extruder auto fan enabled in PWM mode.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4129887/Configuration.zip)

### Steps to Reproduce

1. Select the SKR PRO board : BOARD_BIGTREE_SKR_PRO_V1_1
2. Define SERIAL_PORT 1 end SERIAL_PORT_2 -1
3. Turn on FAN_SOFT_PWM
4. Set E0_AUTO_FAN_PIN to FAN2_PIN and EXTRUDER_AUTO_FAN_SPEED to value lower than 255

**Expected behavior:**
Sucessfull compilation.

**Actual behavior:**
Error during compilation :
[Error.txt](https://github.com/MarlinFirmware/Marlin/files/4129884/Error.txt)

#### Additional Information

Using the same configuration with controller fan does not create compilation error.

#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
#define CONTROLLER_FAN_PIN FAN2_PIN
#define CONTROLLERFAN_SECS 60
#define CONTROLLERFAN_SPEED 63
#endif

boelle, 2020-01-30:
2.0.2 and bugfix are broken right now so hold off for maybe a week or 2

BastR, 2020-01-30:
Duplicate https://github.com/MarlinFirmware/Marlin/issues/15362

boelle, 2020-01-30:
closed as its a dublicate

@zalex94 we only need one open issue for the same problem

closed 2020-01-30 by boelle