MarlinFirmware/Marlin #16722: M117 cannot display : character

jimb15122, 2020-01-29:
### Description

Need a way to disable ":" line concatenation feature for M117 commands.  
Ideally should not require modification of the M117 text.  
e.g. Prefer some form of a prior gcode command, (not compile at time).  
e.g. M??? C0  ; disable colon line concatenation

Another solution would be to ignore :'s that appear inside a quoted string.  However, this method has the disadvantage of requiring modification of the actual M117 command (which may be generated by a slicer that you cannot directly control).
### Steps to Reproduce
M117 PrtTm {print_time}  ; display estimated print time as computed by cura slicer
-- or --
M117 PrtTm 02:22:33

**Expected behavior:** [What you expect to happen]
M117 PrtTm {print_time}  ; display estimated print time as computed by cura slicer
display as: PrtTm 02:22:33

**Actual behavior:** [What actually happens]
displays as: PrtTm 02
#### Additional Information
Confirmed to happen on AnetA8, tronxy x3, ender 3 pro.  Most likely occurs on all systems that support : line concatenation

boelle, 2020-01-29:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

jimb15122, 2020-01-30:
I have attached the factory ender-3 configuration.h as requested, (i am not using a custom build) but ….

In my initial write up, I should have used the term "line splitting" instead of concatenation.
The result I am seeing (and you would see if tested) is entirely expected; the single (in my mind)
**M117 PrtTm 02:22:33**
command is being busted into 3 separate command lines

**M117 PrtTm 02
22
33**

This is STANDARD gcode behavior.  (standard, but little known and unfortunate).  What I am asking for is a way to temporarily disable (at command level, not at custom build time) the line splitting behavior associated with the : character.   

e.g.
M??? C0  ; turn off : separated multi commands in one line support
M117 Estimated Print Time 02:22:33
;M??? C1  ; turn the standard but arcane and dangerous feature back on (or maybe not)

closed 2020-01-30 by jimb15122

ellensp, 2020-01-30:
It displays ":" fine
Tried Marlin bugfix 2.0.x on skr1.3 with RRD 12864 GLCD and RAMPS 1.4 with RRD 20x4 
[image]
[image]

jimb15122, 2020-01-30:
Wow - thanks for the update.  But I am getting different results on 3 different printers with factory builds.  They are all based on 1.1, so that must be why.  Nice to see its fixed in 2.0

Is the above test command coming from an sd card, or a serial port?  I think the : splitting support is only in the SD card command parser code.  ( get_command() in Marlin_main.cpp - search for ':' )   I only use SD based printing, as I tend to keep my PC in sleep mode as much as possible, and windows is too unreliable to keep in the critical path for long prints.

ellensp, 2020-01-30:
Marlin_main.cpp doesn't even exist any more.  

no mention of printing from SD card in original post at all. thus i used serial

From SD.
There do seem to still be issues with ":"  I get lots of errors on console. 
echo:Unknown command: "22"
echo:Unknown command: "33" 
And I only get up to the first ':'  on the disply

ellensp, 2020-01-30:
Its now in queue.cpp 
```
if (card_eof || n == -1 || sd_char == '\n' || sd_char == '\r' || 
              ((sd_char == '#' || sd_char == ':') && !sd_comment_mode
```
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/queue.cpp#L520

AnHardt, 2020-01-30:
#14035

ellensp, 2020-01-30:
I think its a typo from the stone age.. meant to  to be ; to drop any end of line comments that got onto the sdcard.

ellensp, 2020-02-01:
"thinkyhead" has just removed the ":" code as no longer needed.