MarlinFirmware/Marlin #16724: Retagged release 2.0.2 still bad: config directory empty in bugfix_2.0.x, but merged from 2.0.x (release branch)

dok-net, 2020-01-30:
Thanks for removing the src/Marlin.(h|cpp), but more QA was necessary: the updated, probably necessary changes to work correctly, configuration files for all the printers are only in the new Configuration repo, but after the merge for 2.0.2 release the OLD files are again/still in the 2.0.x branch.

## Please check
These files should be looked into if the merge is correct:
Marlin\src\inc\Version.h
Marlin\Version.h
README.md

## Remove as obsoleted by Configuration repo

.github/ISSUE_TEMPLATE/bug_report.md
.github/ISSUE_TEMPLATE/config.yml
.github/ISSUE_TEMPLATE/feature_request.md
.github/labeler.yml
.github/workflows/bump-date.yml
.github/workflows/check-pr.yml
.github/workflows/label-pr.yml
buildroot/bin/backup_ramps
buildroot/bin/build_marlin
buildroot/bin/build_marlin_fail
buildroot/bin/build_marlin_pio
buildroot/bin/build_marlin_teensy35
buildroot/bin/bump_date
buildroot/bin/env_backup
buildroot/bin/env_clean
buildroot/bin/env_restore
buildroot/bin/update_defaults
buildroot/share/atom/auto_build.py
buildroot/share/atom/avrdude.conf
buildroot/share/atom/avrdude_5.10_linux
buildroot/share/atom/avrdude_5.10_macOS
buildroot/share/atom/avrdude_linux.conf
buildroot/share/atom/avrdude_macOS.conf
buildroot/share/atom/create_custom_upload_command_CDC.py
buildroot/share/atom/create_custom_upload_command_DFU.py
config/default/Configuration.h
config/default/Configuration_adv.h
config/examples/3DFabXYZ/Migbot/Configuration.h
config/examples/3DFabXYZ/Migbot/Configuration_adv.h
config/examples/3DFabXYZ/Migbot/Readme.md
config/examples/adafruit/ST7565/Configuration.h
config/examples/ADIMLab/Gantry v1/_Bootscreen.h
config/examples/ADIMLab/Gantry v1/Configuration.h
config/examples/ADIMLab/Gantry v1/Configuration_adv.h
config/examples/ADIMLab/Gantry v2/_Bootscreen.h
config/examples/ADIMLab/Gantry v2/Configuration.h
config/examples/ADIMLab/Gantry v2/Configuration_adv.h
config/examples/AlephObjects/TAZ4/Configuration.h
config/examples/AlephObjects/TAZ4/Configuration_adv.h
config/examples/Alfawise/U20/_Statusscreen.h
config/examples/Alfawise/U20/Configuration.h
config/examples/Alfawise/U20/Configuration_adv.h
config/examples/Alfawise/U20-bltouch/_Statusscreen.h
config/examples/Alfawise/U20-bltouch/Configuration.h
config/examples/Alfawise/U20-bltouch/Configuration_adv.h
config/examples/AliExpress/CL-260/Configuration.h
config/examples/AliExpress/CL-260/README.txt
config/examples/AliExpress/UM2pExt/Configuration.h
config/examples/AliExpress/UM2pExt/Configuration_adv.h
config/examples/Anet/A2/Configuration.h
config/examples/Anet/A2/Configuration_adv.h
config/examples/Anet/A2plus/Configuration.h
config/examples/Anet/A2plus/Configuration_adv.h
config/examples/Anet/A6/_Statusscreen.h
config/examples/Anet/A6/Configuration.h
config/examples/Anet/A6/Configuration_adv.h
config/examples/Anet/A8/_Statusscreen.h
config/examples/Anet/A8/Configuration.h
config/examples/Anet/A8/Configuration_adv.h
config/examples/Anet/A8plus/_Statusscreen.h
[... 293 lines cut ...]
config/examples/Velleman/K8400/Dual-head/README.md
config/examples/Velleman/K8400/Single-head/Configuration.h
config/examples/Velleman/K8400/Single-head/Configuration_adv.h
config/examples/Velleman/K8400/Single-head/README.md
config/examples/VORONDesign/_Bootscreen.h
config/examples/VORONDesign/Configuration.h
config/examples/VORONDesign/Configuration_adv.h
config/examples/Wanhao/Duplicator 6/Configuration.h
config/examples/Wanhao/Duplicator 6/Configuration_adv.h
config/examples/Wanhao/Duplicator i3 2.1/_Bootscreen.h
config/examples/Wanhao/Duplicator i3 2.1/Chippy_Bootscreen.h
config/examples/Wanhao/Duplicator i3 2.1/Configuration.h
config/examples/Wanhao/Duplicator i3 2.1/Configuration_adv.h
config/examples/Wanhao/Duplicator i3 2.1/README.md
config/examples/Wanhao/Duplicator i3 Mini/_Bootscreen.h
config/examples/Wanhao/Duplicator i3 Mini/Configuration.h
config/examples/Wanhao/Duplicator i3 Mini/Configuration_adv.h
config/examples/WASP/PowerWASP/Configuration.h
config/examples/WASP/PowerWASP/Configuration_adv.h
config/examples/WASP/PowerWASP/README.md
config/examples/wt150/Configuration.h
config/examples/wt150/Configuration_adv.h
Marlin/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_ssd_sw_i2c.cpp  under construction

Thanks - then please don't force push, and allow a now release tag 2.0.3?

dok-net, 2020-01-30:
Oh, and while you are at it, could you please tag the Configuration repo as well, that gives better consistency that just the head of, like, `release-2.0.2` branches.

boelle, 2020-01-30:
that would be @thinkyhead that needs to do that, i'm just maintaining the issue list (plenty of work)
so you can see me as a kind of janitor that sweeps the floor (i'm not a coder either)

boelle, 2020-02-08:
@dok-net have you tried bugfix 2.0.x ? that one is updated daily while the release branches(and tags) are not

dok-net, 2020-02-08:
@boelle It's a really really bad idea, turning the whole concept of "releasing" anything under a given version number, to point people toward using unstable development/bugfix branches.
I'm a taking your comment as a justified reminder to close this issues though, because it was fixed (well... there's another issue with how tags and branches are done ;-) ).

closed 2020-02-08 by dok-net

thinkyhead, 2020-02-10:
TL;DR