MarlinFirmware/Marlin #16726: [BUG] BLTouch does not deploy at random UBL Points

kurtisane, 2020-01-30:
Hi,
I'm facing a BUG when using UBL. 
When using a 10x10 Grid for UBL for the first 2 Points BLTouch deploys correctly but at the third Point it just does not deploy. I tried it with the 15x15 Grid and at the 7th or 8th Point the Pin does not deploy. 

I'm on a BTT SKR PRO v1.1 with Marlin 2.0.2. I needed to delete Marlin.h and Marlin.cpp to compile without a error. This is mentioned in a different Issue. Not sure if that is related.

Attach you find the Configuration.h, Configuration_adv.h and the platfromio.ini
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4133042/Marlin.zip)

boelle, 2020-01-30:
2.0.2 and bugfix are broken right now so hold off for maybe a week or 2

italocjs, 2020-01-30:
I had seen some similar issues in the past, in most of them were fault cables, and one in particular (on a printer i made) was being caused by EMI (i was using a single, very thin flat cable to drive the extruder motor, fans and the bltouch), i fixed it adding a 100nf cap between gnd and s, near the bltouch

boelle, 2020-02-02:
@kurtisane have you checked cables etc?

spattinson, 2020-02-03:
You may need to increase the delay, I set this #define BLTOUCH_DELAY 1000 solve my issues, some of them are slower than others, clone etc.

boelle, 2020-02-08:
since we have not heard back from OP we must assume the problem is solved or he have given up

will close this one and we can always reopen once he returns

closed 2020-02-08 by boelle