MarlinFirmware/Marlin #16730: [BUG] FAN1 does not work on SKR PRO

LorenzoSaini, 2020-01-30:
I don't know if it is a bug related to my SKR PRO board, I downloaded the latest marlin 2.0.2 firmware, the issue is than i can't turn on FAN1 by "M106 P1 S255", i duble checked the configuration, configuration_adv and pins_btt_skr_pro_v1_1.h files, in marlin 2.0.1 work as expected.
I don't use Software PWM, Extruder autofan and FAN0 works fine.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4133940/config.zip)

Thanks for the replies.

LorenzoSaini, 2020-01-30:
i have found a solution, in conditionals_post.h.
-Replace
`#define HAS_FAN0 (PIN_EXISTS(FAN))
#define _HAS_FAN(P) (PIN_EXISTS(FAN_##P) && CONTROLLER_FAN_PIN != FAN_##P##_PIN && E0_AUTO_FAN_PIN != FAN_##P##_PIN && E1_AUTO_FAN_PIN != FAN_##P##_PIN && E2_AUTO_FAN_PIN != FAN_##P##_PIN && E3_AUTO_FAN_PIN != FAN_##P##_PIN && E4_AUTO_FAN_PIN != FAN_##P##_PIN && E5_AUTO_FAN_PIN != FAN_##P##_PIN && E6_AUTO_FAN_PIN != FAN_##P##_PIN && E7_AUTO_FAN_PIN != FAN_##P##_PIN)
#define HAS_FAN1 _HAS_FAN(1)
#define HAS_FAN2 _HAS_FAN(2)
#define HAS_FAN3 _HAS_FAN(3)
#define HAS_FAN4 _HAS_FAN(4)
#define HAS_FAN5 _HAS_FAN(5)
#define HAS_FAN6 _HAS_FAN(6)
#define HAS_FAN7 _HAS_FAN(7)
#define HAS_CONTROLLER_FAN (PIN_EXISTS(CONTROLLER_FAN))`
With this From previous build:
`#define HAS_FAN0 (PIN_EXISTS(FAN))
#define HAS_FAN1 (PIN_EXISTS(FAN1) && CONTROLLER_FAN_PIN != FAN1_PIN && E0_AUTO_FAN_PIN != FAN1_PIN && E1_AUTO_FAN_PIN != FAN1_PIN && E2_AUTO_FAN_PIN != FAN1_PIN && E3_AUTO_FAN_PIN != FAN1_PIN && E4_AUTO_FAN_PIN != FAN1_PIN && E5_AUTO_FAN_PIN != FAN1_PIN)
#define HAS_FAN2 (PIN_EXISTS(FAN2) && CONTROLLER_FAN_PIN != FAN2_PIN && E0_AUTO_FAN_PIN != FAN2_PIN && E1_AUTO_FAN_PIN != FAN2_PIN && E2_AUTO_FAN_PIN != FAN2_PIN && E3_AUTO_FAN_PIN != FAN2_PIN && E4_AUTO_FAN_PIN != FAN2_PIN && E5_AUTO_FAN_PIN != FAN2_PIN)
#define HAS_CONTROLLER_FAN (PIN_EXISTS(CONTROLLER_FAN))`

GMagician, 2020-01-30:
Have you checked #16717? It should fix your issue

boelle, 2020-01-30:
and bugfix 2.0.x should be working again, it had a few hours yesterday where it would not compile

LorenzoSaini, 2020-01-30:
Yes, seems that #16717 has fixed the issue, thanks for the report, tomorrow i will test the bugfix branch.

closed 2020-01-30 by LorenzoSaini