MarlinFirmware/Marlin #16733: [FR] Ram Paste extruder and Augur Support

psavva, 2020-01-30:
Ram Extruder and Augur Extruder Support.
I would like to request a feature to support a special type of extrusion system which depends on two motors working together in order to extrude a paste.

The specific printer I have is the CeramBot
https://www.cerambot.com/productinfo/197980.html

However, there are other examples of products with the same extrusion system such as the stoneflower3d system.
https://www.stoneflower3d.com/

### Feature Workflow

The extruder E0 will drive the RAM Extruder.
The extruder E1 should drive the stepper for the Augur Extruder.

When E0 is instructed to extruder the paste, then the Augur which is attached to the nozzle assembly should start mixing the paste to get rid of any air pockets.

They're should be a configuration option which will allow to set a ratio between the E0 and E1 steppers.

To get a good paste extrusion, you will need to have some ratio like 4 revolutions from the augur for every revolution from the Ram extruder motor.

The feature should also allow to set the Augur ratio independently on the LCD Screen and allow for fine adjustments during the print.  The same applies to the Ram extruder motor to be adjusted live during the print.

#### Additional Information
Many people suggest to use the Mixing Extruder on the Forums.  I think that it's not a solution for this problem.

xC0000005, 2020-02-01:
So how is this different than a mixing extruder? With distinct E factors, you could have a high step count for the ram and another count for the extruder, then set a mixing factor of 1:1 which would result in something similar.

I have one of these (the cerambot pro) and would love to understand what the needs are. I haven't had a great deal of luck with mine since my choices are use the driver board which doesn't seem to understand that the ram should only be running when it's extruding and using mixing extruders, where the ram doesn't ever seem to keep up.

psavva, 2020-02-02:
The mixing extruder is designed that 100% of the filament is extruded by using a factors from several extruders. Let's say you setup 2 extruders with a factor of 0.6 and 0.4 which means that 60% of material is pushed through extruder 1 and 40% of material is pushed through extruder 2, giving 100% of the material to be pushed though the hotend.  Sure you could try to setup mixing extruder to give you similar extrusion volume then you have the issue of any changes you try to do from that very 1st setup...  By adjusting the mixing factors live during the print, you're always adjusting both the RAM Extruder and Augur at the same time, always...
You could further try and play around with efactors on each extruder during the print, but then I think you have lost the meaning of what they are designed to do...  You'll never be able to repeatedly get good results in extruding paste because you simply don't have an easy way to do the dine control without manipulation of all extruders...

In the feature I'm proposing, you'll be able to set esteps for your RAM Extruder which should never change.  They're is s formula for that.

For the find turning off the augur bit, you would be able to have fine control on both the extrusion and missing out the clay before it's pushed out of the nozzle.

psavva, 2020-02-02:
You can estimate number of microsteps/mm of extruded material (NS) as follows

Ns = SMs(Dn/De)^2*R/P,

where
 S = number of steps per revolution of the stepper motor (usually 360/1.8 = 200 or 360/0.9 = 400, see specs of the motor)
 Ms = number of microsteps per step (usually 16 or less)
 Dn = nozzle diameter (for example 3mm)
 De = ram extruder (plunger) diameter (i take 50mm)
 R = gear ratio (27 for example)
 P = pitch of the screw in the ram extruder (2mm for example)

20016(3/50)^2*(27/2) = 155.5 microsteps/mm of extruded clay.

Credits: https://wikifactory.cn/+Ceramic3DPrinting/forum/thread/MzAwNTg

psavva, 2020-02-04:
Anyone with some different expert view in this subject?

thinkyhead, 2020-02-04:
Sensible enough. As long as the relationship is linear you can set up an Extruder with any number of "steps-per-unit" that you prefer. If you call the unit a milligram instead of a millimeter, then you just write your G-code to move E in milligrams.

xC0000005, 2020-02-04:
@psavva , if I understand correctly, I think the desired behavior is effectively "paired extruders" in which moving E0 1mm results in 1mm worth of movement in E1, though E0 and E1 don't have the same number of steps per mm. 

Thinking functionally, you'd also want to be able to move E1 independently to prime the ram, and set flow rates for each independently. 

So the result is that a G0 E0 10 would result in 10 x (E0 units per mm) of move interleaved (similar to the mixing extruder functionality) of E1 x (E1 units per mm).

Am I understanding how the functionality would be used?

xC0000005, 2020-02-05:
@psavva, couldn't you double the units per mm for both extruders and set the mix to .5 for both? The result would be the behavior that's wanted, I think.

psavva, 2020-02-05:
Then I would not have fine control over anything if I was to do that.

The other alternative is a hardware solution with 2 external drivers from the same step signal.
One connected to the augur motor, one connected to the extruder motor, setting a different microstep for each, and finally a step generator on the augur to manipulate it manually... But this is something much better suited in the firmware itself...

psavva, 2020-02-05:
Another point is that we would not be able to use mixing extruder for it's intended use, ie mixing of colors, if it where to be utilized to drive the augur in conjunction with a  paste extruder.  That could have some very interesting applications if we are able to use the two features together

psavva, 2020-04-22:
I have found that mixing extruder is not supported with Distinct E-Factors.
I have another FR request open for this to help with this feature definition:
https://github.com/MarlinFirmware/Marlin/issues/17608

closed 2020-06-24 by psavva