MarlinFirmware/Marlin #16734: bltouch issue

yoc1983, 2020-01-30:
After adding bltouch there's a weired problem,
everything seems to work but..
after running G29 the prob will test 6 point and then stop
i configured it to do a 3x3 mesh
if i am running G29 V1 i see in terminal the prob numbering 1/9, 2/9 ... then 7/9
but stopping in the 6th position .
bed size set to 235x235

ender 3 pro, skr 1.3, bltouch 3.0 original,

boelle, 2020-01-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

yoc1983, 2020-01-31:
@boelle thanks mate here it is:
[joe-config.zip](https://github.com/MarlinFirmware/Marlin/files/4136909/joe-config.zip)

opiswahn, 2020-01-31:
I had that once... try disabling Bilinear_Subdivision, that helped me

yoc1983, 2020-01-31:
@opiswahn 
tried now, i enabled linear instead.. same thing stopping after the 6th point
also looks like he didn't saved the values
[image]
this is running M420 after bilinear:
[image]

yoc1983, 2020-01-31:
OK i gave a manual direction and finally managed to sample the all 3x3 
G29 X3 Y3 L50 R150 F50 B150 T V4
[image]
that's nice..see how i go from here thanks @opiswahn

sjasonsmith, 2020-02-01:
Can you verify the approximate date of the bugfix-2.0.x code you are using. Is it code you just downloaded, or something older. There were a lot of issues like this a month or so ago that were resolved, so I want to find out if this is possibly those same issues.

yoc1983, 2020-02-01:
@sjasonsmith OMG.. bless you Jason
there was a human bug
i used the reconfigured marlin from here:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3
guess it was not updated
i downloaded now a fresh copy of marlin and configured everything
and it's working perfectly.
sorry for the trouble guys i am a beginner with this. 
thanks for the help

boelle, 2020-02-01:
@yoc1983 so we can close this one?

yoc1983, 2020-02-01:
@boelle 
yes. this bug was fixed
thanks

closed 2020-02-01 by boelle