MarlinFirmware/Marlin #16736: [BUG] Auto Home on Delta not working

jamilbn, 2020-01-31:
### Bug Description

On Marlin 2.0.1 everything was working. Auto home, Auto Calibration.
With Marlin 2.0.2 the auto home moved only one tower at a time and that was a problem since the effector start to hit that tower.
With Marlin 2.0.3 same thing.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4137682/Configurations.zip)

### Steps to Reproduce

1. Just select Auto Home or start the Auto Calibration.

**Expected behavior:** [The printer will Auto Home moving the 3 towers together]

**Actual behavior:** [It will move only one tower, if that tower is close to the endstop it will move the second but not the third tower.]

thinkyhead, 2020-01-31:
Do you still have the configs you were using with 2.0.1? Please post those too so I can quickly check that all the settings match up ok.

Have you done M502, M500 to clear out old EEPROM data?

What does `M119` say about your endstops? Are they showing up as triggered / open appropriately, or do they show up as always triggered?

sjasonsmith, 2020-01-31:
I just reproduced it on my printer. The problem goes away if I reverse commit  2325bede8a079845dd19ce8020350c1d8f8a18f1, which altered pin assignments for the Trigorilla 1.4.

Now I'm trying to find out how to verify that my board actually is a Trigorilla 1.4. I just assumed it was because that is what was already in the Anycubic Kossell LP example.

sjasonsmith, 2020-01-31:
I can't find any information to help identify a board version other then a comment from @thinkyhead in #11356 indicating that a 1.4 has support for two extruders. In that case my board would be a 1.4.

sjasonsmith, 2020-01-31:
After some searching I have found two YouTube videos which show the motherboard for the AnyCubic 4Max Pro, which is the printer which inspired the pins file change.

Both of them show a daughterboard connecting to all the pins on one end of the motherboard, which presumably re-allocates them in a way that works for this printer. I have circled the location of the endstop connectors in red, and you can see they have been all routed to the daughter board.

Given this, it seems like the change to the Trigorilla 1.4 file was an error, and this printer would need to be supported with either a new pins file, or by providing a way to override the default Trigorilla 1.4 behavior.

[image]

jamilbn, 2020-01-31:
> Do you still have the configs you were using with 2.0.1? Please post those too so I can quickly check that all the settings match up ok.

Used the configs provided by marlin in all the three cases.

> Have you done M502, M500 to clear out old EEPROM data?

Only using the LCD by choosing Initialize EEPROM. Always use that.

> What does `M119` say about your endstops? Are they showing up as triggered / open appropriately, or do they show up as always triggered?

Not near the print right not to test. But with what firmware do you want me to test?

[Configs 2.0.1.zip](https://github.com/MarlinFirmware/Marlin/files/4139490/Configs.2.0.1.zip)

jamilbn, 2020-01-31:
> I just reproduced it on my printer. The problem goes away if I reverse commit [2325bed](https://github.com/MarlinFirmware/Marlin/commit/2325bede8a079845dd19ce8020350c1d8f8a18f1), which altered pin assignments for the Trigorilla 1.4.
> 
> Now I'm trying to find out how to verify that my board actually is a Trigorilla 1.4. I just assumed it was because that is what was already in the Anycubic Kossell LP example.

With 2.0.1 when I auto home all the towers move at the same time to the endstop switches.
With 2.0.2 and 2.0.3 it first try to move one tower. If that tower isnt near the height to reach the switches it will just crash the effector on the tower.

sjasonsmith, 2020-02-01:
@jamilbn, this should now be fixed in the bugfix-2.0.x branch. Could you test it, and close this issue if it is resolved?

closed 2020-02-01 by thinkyhead