MarlinFirmware/Marlin #16739: Delta eight

Riciweb1004, 2020-01-31:
height show in Configuration>Delta calibration>Delta settings  is not real eight but the 
eight - Probe_z_offset

My machine is Anycubic delta linear plus

[image]
[image]
[Marlin_any_kossel_plus.zip](https://github.com/MarlinFirmware/Marlin/files/4139283/Marlin_any_kossel_plus.zip)

sjasonsmith, 2020-02-01:
I can confirm this is true, and that the same number are reported at the console by the G33 command, and reported back through M503.

The printer behaves correctly, though, allowing movement through the full range of motion.
For my printer, it reports a height of 277.18, but my actual travel limits are 0-294, which corresponds to the offset of my Z probe.

It is hard for me to know if there is an issue here.
Yes, the height is wrong if you measure it and enter it manually, but if you are doing that you probably do not have a probe offset.

This might be more intuitive if changed, but would impact all existing delta configurations with probes.

thinkyhead, 2020-02-01:
Maybe @LVD-AC can offer some insight into the display and why the height is expressed this way. Perhaps it needs to be displayed with the offset included.

LVD-AC, 2020-02-01:
Delta Height in Marlin is (now) defined as the distance between the trigger point of the endstops at the top and the trigger point of the probe at the bottom or in absence of a probe the point were the nozzle grabs a piece of paper ; you can call this a manual probe. And no probe means also no probe offsets, so only on deltas with no probe the delta height corresponds to the maximum travel distance.

> Perhaps it needs to be displayed with the offset included.

No

The internal variable is displayed. It is only used twice for calculations (G28 homing and G33 calibrating); all the rest is user interaction. So for consistency you would also need to include the offset in all other places like setting the delta height with the dispay or with M665, reporting the delta height by M503 and G33, storing and reading it to EEPROM with M501 (slicer programs can also display/change the delta height), reading it from Configuration.h, ... and the list goes on and on and on.

sjasonsmith, 2020-02-07:
So it seems like this is entirely expected behavior, and this issue can be closed?

Riciweb1004, 2020-02-08:
> The internal variable is displayed. It is only used twice for calculations (G28 homing and G33 calibrating); all the rest is user interaction. So for consistency you would also need to include the [...]

I don't agree, the display must show the actual height, the probe offset is an internal variable that is only needed during the calibration phase, the height displayed in the delta setting is wrong and misleading, the slicers take into account the actual height and not the height without offset!

sjasonsmith, 2020-02-08:
@LVD-AC, has it always been stored and displayed this way, or was that a recent change?

Riciweb1004, 2020-02-08:
it's a change that has appeared now

LVD-AC, 2020-02-08:
@sjasonsmith this was changed in 1.1.6 (october 2017); G33 was remidiated accordigly in 1.1.7 but the delta height is basicly used in G28 and that caused problems that were only fixed after the final release of 1.1.9 (december 2018). So between those dates there was a big problem on deltas:

You needed to recalibrate the delta height and home the printer to that new height after each offset change in order to make the printer work correctly. In 1.1.7 a temporay solution was offered in G33, to recalibrate the delta height in accordance with a changed offset without probing, just rehoming (G33 P0).

With the finally adopted solution offset changes take immediate effect (babystepping during printing works now on deltas with a probe) and the homing works correctly without having to recalibrate all the time.

So you are all right : @sjasonsmith the behavior is intended; @Riciweb1004 to the user it is indeed pretty confusing (but at least it works now); and @thinkyhead indeed maybe we should reconsider.

It makes much more sense to use, calibrate to, display, report and store the much more familiar value of delta height. So basicly include Z_PROBE_OFFSET_FROM_EXTRUDER into the delta_height variable and home to the additional offset set by M212 instead of homing to the total offset set by M851.

This requires no changes to G33 or any place else at all, only the 3 instances in G28: 

`delta_height - zprobe_zoffset`

should be changed to

`delta_height - (zprobe_zoffset - Z_PROBE_OFFSET_FROM_EXTRUDER)`

Riciweb1004, 2020-02-10:
@LVD-AC Thanks for telling the story of delta homing up to this version.
Out of laziness, I have always been satisfied with the stock FW, until I set up the marlin 1.1.8 for the first time, but I was not satisfied with the calibration and I was backward because it seemed the lesser evil...
`babystepping during printing works now on deltas with a probe`
I tried to enable it, but on my Delta Anycubic with Trigorilla, the printer hangs and asks for a reset, so I would say that there is still some problem here ...

Thanks for taking the time to discuss this topic :-)

boelle, 2020-06-20:
@Riciweb1004 is this one still an issue?

boelle, 2020-06-22:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-22 by boelle