MarlinFirmware/Marlin #16740: [FR] Idea for Dual Z homing with BLTouch

IteDas, 2020-01-31:
I recently upgraded my Ender to a dual z Axis (driven via 2 drivers). I also have a BLTouch. 

It is a common problem that one of the steppers might be (by accident, by itself or by someone applying force to the Torque Screw) moved hence you end up with a skew on the bed. To mitigate that you could probe the two ends of the bed where the motors are mounted as follows to make sure they have not drifted while the devices was powered off:

**Process** 

- G28 (or may be G29) would home with BLTouch not in the centre but on the defined Bed Position in Marlin (where first Z Motor is mounted)
- The position would be stored
- Z Axis would move down Xmm (to avoid hitting the bed) 
- Print Head would move to opposing side and probe again.
- The difference in both measurements would then be applied to the lower Z Motor to level them

After that a normal G28 could be issued to home normally. Also the previous process might have to be repeated X times as moving on side up affects the other side as well.

InsanityAutomation, 2020-01-31:
See G34 in configuration adv.

IteDas, 2020-01-31:
I'm sorry, I really did not find that command. Thank you.

closed 2020-01-31 by IteDas