MarlinFirmware/Marlin #16744: [BUG] PT100 + SKR1.3 + TEMP_SENSOR_0 21 = 1ºc

sn4k3, 2020-01-31:
### Bug Description

Before this i was using `TEMP_SENSOR_0 20` with PT100 at 3.3V VCC and it works, now with new marlin update i change to TEMP_SENSOR_0 21 and the result is 1ºc at hotend.

### My Configurations

`#define TEMP_SENSOR_0 21`

#### Additional Information: Wiring

My wiring: https://cdn.thingiverse.com/assets/ef/59/cc/a9/fc/PT100_Wire.png
Alternate wiring with same result: https://cdn.thingiverse.com/assets/2a/cd/e0/96/d7/PT100_WireAlt2.png

Please note that i removed R18/R19 and C18/C19 from motherboard
The PT100 Signal goes to TH0 + (0.24)

But connections are ok since it work fine when using `#define TEMP_SENSOR_0 20` despite the 3.3v logic/vcc

boelle, 2020-01-31:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

sn4k3, 2020-01-31:
Here: 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4141447/Configuration.zip)

atoomnetmarc, 2020-02-01:
@sn4k3 If I understand sensor type 21 correctly, that will only work when the amplifier board is powered by exactly 5.000V and the CPU by exactly 3.300V.
You have it powered by 3.3V, which is fine, and safe. Sensor type 20 works correctly when the power supply of the amplifier board is the same as the reference voltage of the ADC in the CPU, which for a SKR1.3 is.

I suspect when powering the amplifier board by 5V that disconnecting the sensor would result in 5V in the output and instantly fry the CPU. The ADC's in the CPU off the SKR1.3 are not 5V tolerant.
Keep the amplifier board on 3.3V.

thinkyhead, 2020-02-01:
Use 20.

closed 2020-02-01 by thinkyhead

sn4k3, 2020-02-01:
Ah, the comment about 21 lead me to think that would be the way to go with my setup, i think it should be corrected, some thing like:
*    21 : Pt100 with **5V feed** circuit in the Ultimainboard V2.x with 3.3v excitation (STM32 \ LPC176x....)
Thank you

thinkyhead, 2020-02-01:
We're not sure 21 should stick around at all. It's under discussion.