MarlinFirmware/Marlin #16746: SHOW_TEMP_ADC_VALUES are not in line with thermistor_1.h

rahaymore, 2020-02-01:
### Bug Description
Recv:  T:89.53 /0.00 (2674.00) B:22.81 /0.00 (3922.00) @:0 B@:0
Recv:  T:88.58 /0.00 (2702.00) B:22.81 /0.00 (3922.00) @:0 B@:0

Values range from 3500 @ 31c to 350 @ 210c
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4142034/Configuration.zip)

### My Configurations
Ender 3 with stock hotend
SKR v1.3 with #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
#define TEMP_SENSOR_0 1

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Unzip fresh download of bugfix-2.0.x (also tried all released versions of 2.0)
2. Configure configuration.h for SKR V1.3
3. Enable SHOW_TEMP_ADC_VALUES in configuration.adv.h
4. Update firmware and monitor terminal

**Expected behavior:** 
SHOW_TEMP_ADC_VALUES to be in the range of thermistor_1.h values

**Actual behavior:** 
Values are 3 times greater than thermistor_1.h values at 25c.

#### Additional Information
I also have a Ender 5 with SKR mini E3 V1.2 motherboard which works as expected.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ManuelMcLure, 2020-02-01:
The values in the thermistor table are normalized to 10 bits, i.e. 0-1023. The ADC produces 12 bits (0-4095) so the values are divided by 4 (not 3) to look them up in the thermistor table. This is not a bug.

rahaymore, 2020-02-01:
Thank you.
Could you explain why the SKR mini E3 produces 10 bit number?

ManuelMcLure, 2020-02-01:
Because different boards have different ADC hardware. This is handled by the HAL - if you look at the ``HAL.h`` file for your board it specifies what the resolution is as ``HAL_ADC_RESOLUTION``. LPC176X and SAMD51 are 12 bits, all other boards are 10.

closed 2020-02-01 by thinkyhead