MarlinFirmware/Marlin #16748: [BUG] Vertical fine artifacts

GarthSnyder, 2020-02-01:
### Bug Description

I hesitate to call this a bug, and in fact it's perilously close to one of the endless "hey, what's wrong with my print" threads. The only reason I'm filing this is that I've been battling this particular print artifact for a couple of months with no success, but to my surprise, installing Klipper seems to have resolved it completely. Is it possible that Marlin is doing something to bring out this issue?

The attached video shows two prints from identical G-code, Marlin on the left and a vanilla install of Klipper on the right. As you can see, the Marlin print has repeated vertical striations and the Klipper print does not.

* The artifacts are regularly spaced and appear on all prints. However, they are subtle and overtly visible only on prints made with shiny plastic under specular lighting. You cannot feel the ridges with a finger.

* Artifacts are always perfectly vertical. There is no variation with print geometry or curvature.

* They are completely independent of print speed, acceleration, and print geometry. This is not ghosting or vibration.

* The frequency is regular. The artifacts appear on walls oriented parallel to both X and Y axes. Walls at an angle show the same pattern, but the frequency varies slightly with angle.

* The frequency is too high and too regular to be related to a bearing or a rough patch on a wheel or track. Nevertheless, all of those components have been scrutinized in detail for flaws.

* The artifacts are independent of microstepping, Trinamic mode (StealthChop vs. SpreadCycle), microstepping interpolation (MicroPlyer), and motor current.

* The artifacts are independent of linear advance and junction deviation. 

* The artifacts are independent of Marlin settings SQUARE_WAVE_STEPPING and MINIMUM_STEPPER_PULSE.

* I have replaced the printer's mainboard, stepper drivers, and stepper motors, with no change. Original mainboard is a clone of the MKS Gen L with cloned TMC2100 stepper drivers. New board is an SKR V1.3 with genuine TMC2208 drivers. For motors, I went to 0.9 degree OWCs after reading a thread on the Prusa forums where Guy Kuo reports a lot of interesting data about vertical striations on MK3s (he calls these "vertical fine artifacts"). However, since 0.9 degree motors made no difference (or actually worsened the problem), I suspect that my artifacts may be different from those on the MK3.

* I didn't analyze in as much detail as Guy Kuo, but his conclusion was that the pattern recurred every four full stepper increments. My pattern spacing is similar. Broadly speaking, this type of pattern seems to be associated with nonlinearity in motor rotation.

### My Configurations

[Configuration.h](https://gist.github.com/GarthSnyder/56b6256253b5a2f158dd06e9c65eada9)
[Configuration_adv.h](https://gist.github.com/GarthSnyder/1d5d21b9ecdc3d8c98de2e7aede40e4c)
[printer.cfg](https://gist.github.com/GarthSnyder/c3143e3560d394dc97871f646b944ad3) (Klipper configuration)

The hardware is an Artillery Sidewinder X1, version 4. Now with BigTreeTech SKR V1.3, TMC2208 stepper drivers, and stepperonline.com 17HM19-2004 stepper motors on X and Y. Also has BLTouch.

### Steps to Reproduce

Print a box.

**Expected behavior:** 

A box.

**Actual behavior:**

A box with regularly spaced vertical lines running up the sides.

#### Additional Information

[Marlin vs. Klipper - HD 720p.mov.zip](https://github.com/MarlinFirmware/Marlin/files/4142217/Marlin.vs.Klipper.-.HD.720p.mov.zip)

Just to be clear, I am not requesting or expecting technical support on this. But it does seem possible that the problem originates somewhere in the bowels of Marlin's motor control system. I would be happy to provide additional information if you're interested in looking into this.

Lord-Quake, 2020-02-01:
Could you post the G-code. I'll try and see how the results are with my printers, just for comparison sakes.

thinkyhead, 2020-02-01:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-02-01 by thinkyhead

GarthSnyder, 2020-02-01:
Just as a point of idle curiosity, did you actually read this bug report, @thinkyhead?

sedlacekdavid, 2020-02-25:
Reporting this on TMC2209 & FW Marlin 2.0.3

If klipper solves this, I'd consider moving to Klipper as this is the only solution currently ??

sedlacekdavid, 2020-02-25:
@GarthSnyder  I can't believe how much different it is. The thing is recently I've spent a lot of time tuning my printer to get appropriate results - linear rails etc. But I just couldn't get rid of those vertical artifacts no matter what. So I'll definitely test Klipper on my printer.

thinkyhead, 2020-02-25:
> Just as a point of idle curiosity, did you actually read this bug report…?

Yes, but there's simply nothing actionable here! We need a specific bug and something that can be pointed to as being wrong or inefficient or…. This PR is simply proposing "make Marlin better" which is what we're already trying to do every day.

If someone really wants to delve into this particular issue and make Marlin's stepping engine work as smoothly as Klipper that would be just great! The best we have at the moment is `ADAPTIVE_STEP_SMOOTHING` to reduce aliasing to the minimum using as much processor power as is available.

I personally don't have time to get into rewriting the stepper engine in Marlin right now because I'm dealing with the day to day of trying to follow up on hundreds of issues, fulfill a huge client project, and then steal enough time to work on the new configuration tools.

GarthSnyder, 2020-02-25:
@thinkyhead, a reproducible test case is by definition actionable, should someone want to investigate it. It's perfectly fine if there's no interest. It's just a bit strange to have an issue report dismissed as a support request, especially when it says "this is not a support request" (or words to that effect) at both the beginning and end of the report, and especially when the submitter doesn't use your software anymore. I can only offer you the feedback that I came away from this exchange thinking, "Hmm, I guess that's the last potential Marlin issue I'll be bothering to write up."

@sedlacekdavid, in my case the ultimate cause was belt noise, probably from a random combination of cloned 2GT pulleys, idlers, and belts from different manufacturers. The clones often don't reproduce the exact tooth profile of the original (Gates) equipment, so there's a small wiggle on every tooth engagement.

Klipper hides the issue better than Marlin because its stepper timing seems to be smoother and more consistent, but it doesn't solve the underlying issue. So as @thinkyhead says, there ultimately isn't much to do here other than "make Marlin smoother".

The exact spacing of vertical artifacts can help to disambiguate belt issues from true "vertical fine artifacts", which recur per four full stepper cycles. On my printer (Sidewinder X1), belt noise is about 1.8mm/cycle while VFAs would be more like 1.3mm/cycle. But these will vary depending on your exact hardware and ratios.

sedlacekdavid, 2020-02-26:
@GarthSnyder So you think it's just a timing belt issue? Looking at your photo comparing the two, I don't think so...

sedlacekdavid, 2020-02-26:
I'll try ADAPTIVE_STEP_SMOOTHING, thanks @thinkyhead

thinkyhead, 2020-02-26:
> It's perfectly fine if there's no interest.

Everyone wants it to work better, but this is something we are always working on. There is no *special action* to take regarding this particular problem. It will shake out from our general optimizations, or someone with a big brain will come along and revolutionize the codebase, or there will be a sprint on it under someone's initiative.

This is not the place to explore the question because we want to use this issue queue to manage issues, not to have long discussions. If you personally want to get down with developers and smart geeks, the better place is not here but the Discord server. Tally-ho! Good luck! I wish I personally had the time right now.