MarlinFirmware/Marlin #16753: [BUG] SKR E3 DIP Extruder motor never turns off once energized

opcow, 2020-02-01:
### Bug Description

The E stepper doesn't disable after it has been enabled.

I noticed a couple of hours after a print had finished that the extruder stepper was still on.  Neither using the LCD's disable steppers command, Octoprint's motors off button, or sending the M18 command to the printer would turn off the stepper. Power cycling (or causing a reboot using `M997`) was the only way to get the E motor turned off.

### My Configurations

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4142575/configs.zip)

### Steps to Reproduce

1. Issue command `M17 E`.
2. Check that extruder stepper motor is energized.
3. Issue command `M18 E`
4. Check that extruder stepper motor is de-energized.

**Expected behavior:**
Stepper will turn off after the `M18 E` command.

**Actual behavior:**
Stepper remains energized after the `M18 E` command.

#### Additional Information
The board is an SKR E3 DIP (STM32F103RC). I'm using TMC2208 drivers on XYZ and a DRV8825 on E. If I build 2.0.2 using the same configuration the problem does not exist.

I tried using `M18 E0` and `M18 E1` commands as well. Same result.

thinkyhead, 2020-02-01:
- Every time you print steppers won't turn off?
- Every time you turn a stepper on it won't turn off?

opcow, 2020-02-01:
Correct. If the stepper motor ever turns on it will not turn off again until I power cycle the machine.

opcow, 2020-02-01:
I switched out the DRV8825 for a TMC2208 and reconfigured for that. Same issue. The E stepper will never turn off once it has been turned on.

thinkyhead, 2020-02-01:
Very interesting…. I did a diff and looked for relevant changes between 2.0.2 and 2.0.3 and nothing jumped out. I was thinking perhaps the disable for extruders was broken. But I see nothing unusual going on in the firmware.

Here is something you can try, but use caution! … Exchange the pin numbers of the E0_STEP, DIR, and EN pins with those of X STEP,DIR,EN pins in the E3 DIP pins file and re-build. Now, don't do anything except the M18 to turn the stepper on and off. See if the X stepper now turns on and off when you tell the E stepper to turn on and off. As soon as you get the results, flash it back the old way.

Use the latest bugfix because it's had a few patches today and maybe one of them will have an effect.

I will continue to look at the code that disables E steppers and see if anything looks weird.

One other thing you can do is to enable `PINS_DEBUGGING` and use M43 and get a pins report. This will tell at least tell us if we have pins conflicting someplace.

I'll continue to keep my eyes open for some recent change that looks suspicious.

opcow, 2020-02-02:
OK, did the pin swapping. The X stepper turns on with `M17 E`, but it won't turn off. So same behavior except on X instead of E.

dongl1, 2020-02-02:
I'm having the same issues with 2.0.3 on a ReArm with ramps 1.6 as well. When I switch back to 2.0.2 everything works as it should. I've attached my configuration files. I am running dual extruders and the issue is there for both steppers.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4144347/Configuration.zip)

thinkyhead, 2020-02-04:
Thanks for the extra info. That implies it is not platform-specific but that something else is going on. I am checking through all the changes right now and hopefully it will be something obvious.

thinkyhead, 2020-02-04:
Problem solved. A rather foolish typo caused this, sorry!

closed 2020-02-04 by thinkyhead

opcow, 2020-02-04:
Yup, that fixed it.

commit 15b6ad41 (2020-02-04): Fix E stepper stays on bug

Fixes #16753