MarlinFirmware/Marlin #16754: Got error when complies

Talha909, 2020-02-01:
I had downloaded the 2.3. After compile i got two errors. 

`
{
	"resource": "/e:/Marlin-2.0.x/Marlin/src/lcd/menu/menu.h",
	"owner": "cpp",
	"severity": 4,
	"message": "'_thisItemNr' may be used uninitialized in this function [-Wmaybe-uninitialized]",
	"startLineNumber": 342,
	"startColumn": 37,
	"endLineNumber": 342,
	"endColumn": 37
}

In file included from Marlin\src\lcd\menu\menu_temperature.cpp:31:
Marlin\src\lcd\menu\menu.h: In function 'void menu_temperature()':
Marlin\src\lcd\menu\menu.h:342:37: warning: '_thisItemNr' may be used uninitialized in this function [-Wmaybe-uninitialized]
 #define END_SCREEN() } screen_items = _thisItemNr
                                     ^
Marlin\src\lcd\menu\menu.h:326:40: note: '_thisItemNr' was declared here
   int8_t _menuLineNr = encoderTopLine, _thisItemNr; \
                                        ^~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:339:22: note: in expansion of macro 'SCREEN_OR_MENU_LOOP'
Compiling .pio\build\LPC1769\src\src\libs\least_squares_fit.cpp.o
 #define START_MENU() SCREEN_OR_MENU_LOOP(true)
                      ^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_temperature.cpp:165:3: note: in expansion of macro 'START_MENU'
   START_MENU();`

boelle, 2020-02-01:
i see a warning, does it compile with a fail?

boelle, 2020-02-01:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

sjasonsmith, 2020-02-01:
I also noticed this warning yesterday. Seems like a new one, I don’t remember seeing it before.

thinkyhead, 2020-02-01:
It's not a problem. The variable is always initialized when it needs to be.

closed 2020-02-01 by thinkyhead