MarlinFirmware/Marlin #16758: [FR] PSU STARTUP/BOOT DELAY

tiagosampaio79, 2020-02-02:
Hello,
  Last week, I compiled (Platformio) Marlin 2.0.2 firmware to an MKS SGEN_L (LPC1768), with PSU_CONTROL enabled so that I could turn the control board off with M81 in End-GCode, and a relay module connected to a servo PIN (1_23). However, I realized that, right after M81 turns the relay off, the board reboots after a few seconds and turns the relay on again, preventing the shutdown. I think it happens because the LPC1768 can somehow manage power and get some "stamina" from capacitors while shutting down. So I added these two lines, which fixed the problem:

Marlin.cpp (238): delay(PSU_STARTUP_DELAY);
Configuration.h (327): #define PSU_STARTUP_DELAY 20000

I am not sure that it's the right way to fix the problem, and I appreciate any further suggestions.
Ps.: This is my first time on GitHub, so if it's not the right place to post the request/issue, I can correct it. Sorry.

italocjs, 2020-02-03:
your issue seems like EMI, i had a lot of these in the past on some boards we design,  to help fix this you might want to use a relay with dual optocoupling or a SSR to avoid sparks at all, 

you might also want to use a pullup or puldown external resistor and "set" the relay logic, so when there is no power to the board, the resistor pulls it to GND and the relays stays open, until the board boots and set the pin to HIGH.  as you pointed, the board does "stay on" while there is still some juice in the capacitors (usually less than a second),

tiagosampaio79, 2020-02-03:
We tried a pulldown resistor, but it didn't work as well. The SGEN_L Servo signal is 3.3v (4mA?), not enough to toggle a Fotek SSR-25DA, although we are using a dual optocoupler relay, which is 20cm away from the mainboard.

italocjs, 2020-02-03:
At least for my understanding, the current from the board will only turn on the optocoupler led (which essentialy is a transistor with a gate triggered by light), so as long as the board provides the minimum current for the optocoupler (10ma top i guess) your should be fine using this relay or a ssr triggered by the optocoupler.

for the resistor, i have been using 10k here without any problems (but my printers are based on atmega2560)

I'll soon be writing something to help me shutdown the raspberry safely (triggering a pin to shut it down and wait until the raspberry confirm the "safe to shutdown" status on another pin, to only then shut the atx psu off), and i guess ill start running in those kind of issues too,

tiagosampaio79, 2020-02-03:
> At least for my understanding, the current from the board will only turn on the optocoupler led (which essentialy is a transistor with a gate triggered by light), so as long as the board provides [...]

YES!! You are totally right. I couldn't enable the SSR through servo pin 1_23, so I used a PNP transistor controlled by that pin (3.3v 4mA), using servo's VCC and GND and it worked flawlessly, without any need to delay during boot time. So it really was EMI.

   Thanks in advance!

closed 2020-06-30 by github-actions[bot]