MarlinFirmware/Marlin #16760: [BUG] Unable to compile BOARD_MKS_ROBIN_LITE, current bugfix 2.0.x

ellensp, 2020-02-02:
### Bug Description

With default Configuration files.
update MOTHERBOARD BOARD_MKS_ROBIN_LITE
and set default_envs = mks_robin_lite
This errors with a  'CYCLES_PER_MICROSECOND' was not declared in this scope.

### My Configurations

[Configs_MKS_ROBIN_LITE.zip](https://github.com/MarlinFirmware/Marlin/files/4143642/Configs_MKS_ROBIN_LITE.zip)

### Steps to Reproduce

1. git clone https://github.com/MarlinFirmware/Marlin.git
2. open vscode and open marlin src directory
3. change to bugfix-2.0.x branch
4. edit platformio.ini, change default_envs to mks_robin_lite
5. edit Configuration.h, change MOTHERBOARD  to BOARD_MKS_ROBIN_LITE
6. update all libraries, boards, platforms in PlatformIO
7. Click PlatformIO Build.

**Expected behavior:** 

Expect it to compile.

**Actual behavior:** 

Fails to compile with the following error

```
In file included from /home/user/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/wirish/boards.cpp:48:0:
/home/user/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/wirish/boards.cpp: In function 'void init()':
/home/user/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/boards.h:121:49: error: 'CYCLES_PER_MICROSECOND' was not declared in this scope
 #define SYSTICK_RELOAD_VAL              (1000 * CYCLES_PER_MICROSECOND - 1)
                                                 ^
```

sjasonsmith, 2020-02-02:
I didn't see this error when I tried to build your configs.
I did need to change `SERIAL_PORT` from 0 to -1, but after making that change it built just fine.

ellensp, 2020-02-02:
I knew there would be other errors... once it got past this point... wasn't worried about those.

Can you compare my versions with yours? 
Full build log https://pastebin.com/Qz17KgHG

ellensp, 2020-02-02:
Ok, I removed my .platformio  folder. and forced it to reinstall
now it compiles (till the serial error) 

the difference, the order of the packages.

Fails
< PACKAGES: framework-arduinoststm32-maple 1.10000.190819 (1.0.0), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)

Works
> PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0)

closed 2020-02-02 by ellensp