MarlinFirmware/Marlin #16761: [BUG] compiling issue from 2.0.1 to 2.0.3 ftdi EVE

TB1405, 2020-02-02:
Compiling isnt possible with #define EXTENSIBLE_UI. 
With the version 2.0.0 and the TouchUI folder from fysetc it was possible to compile successfully and the display works.

I have now adapted the touchUIfolder of fysetc, for marlin 2.0.3 but with both folders of fysetc or marlin I get an error message at the end of compiling  https://pastebin.com/9Mna1s1g
in the release from 2.0.0 to 2.0.1 was changed in M0_M1.cpp
#include "../../inc/MarlinConfig.h" to #include "../../inc/MarlinConfigPre.h" and in this file i think is the error.
VisualStudio says   it cant open this file `#include <stdint.h>` . 
i uploaded the complete marlin file because i changed a few things in touchUI  folder 
but the same error message appears with a fresh download 2.0.3 and enabled EXTENSIBLE_UI

### My Configurations
[Marlin-2.0.3.zip](https://github.com/MarlinFirmware/Marlin/files/4143818/Marlin-2.0.3.zip)

### Steps to Reproduce
download the marlin file add it to VScode and try to compile

TB1405, 2020-02-02:
alright was a problem of the old Fysetc libary. 
We only need to add
  #undef max
  #define max(a,b) ((a)>(b)?(a):(b))
  #undef min
  #define min(a,b) ((a)<(b)?(a):(b))

in the compat.h file in the the touchUI folder

boelle, 2020-02-08:
@TB1405 if you have the skills then a PR is the fastest way to get it fixed

@thinkyhead does the proposed fix make any sense?

TB1405, 2020-02-11:
GerogeFu  has started a pull request. 
Fix is included here 
https://github.com/MarlinFirmware/Marlin/pull/16830/commits/7cef4f52a94f57b4f8e88cb4a95d55a7235aa59c

TB1405, 2020-02-15:
Should work now for stm32

closed 2020-02-15 by TB1405