MarlinFirmware/Marlin #16762: [BUG] Compiling issue "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."

kramer04, 2020-02-02:
Bug Description

define E0_AUTO_FAN_PIN PA8 triggers compiling issues

`In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL.cpp:30:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:1444:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
        ^~~~~
Compiling .pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/onboard_sd.cpp.o
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:1444:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
        ^~~~~
Compiling .pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/persistent_store_eeprom.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/persistent_store_flash.cpp.o
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/Servo.cpp:26:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:1444:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
        ^~~~~
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/HAL.cpp.o] Error 1
In file included from Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/dogm/u8g_com_stm32duino_fsmc.cpp:29:
Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/SanityCheck.h:1444:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
        ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/dogm/u8g_com_stm32duino_swspi.cpp:21:
Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/SanityCheck.h:1444:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
        ^~~~~
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/HAL_SPI.cpp.o] Error 1
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/onboard_sd.cpp:15:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:1444:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
        ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/persistent_store_eeprom.cpp:23:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:1444:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
        ^~~~~
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/Servo.cpp.o] Error 1
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/persistent_store_flash.cpp:32:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:1444:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
        ^~~~~
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/dogm/u8g_com_stm32duino_fsmc.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/dogm/u8g_com_stm32duino_swspi.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/persistent_store_eeprom.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/onboard_sd.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/persistent_store_flash.cpp.o] Error 1`

### My Configurations
Ender3 + SKR MINI E3 V1.2 + TouchMi

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4143954/configuration.zip)

### Steps to Reproduce

#define E0_AUTO_FAN_PIN PA8

boelle, 2020-02-02:
have you tried to set E0_AUTO_FAN_PIN to something else than PA8 ?

ellensp, 2020-02-02:
In pins_BTT_SKR_MINI_E3.h is #define FAN_PIN    PA8

There is only one software controllable fan on this board Fan0 on PA8. It must be assigned to FAN_PIN and used for part cooling.
Fan1 is always on and should be used for hotend cooling.

boelle, 2020-02-02:
so the fix mus be to change the pins file and set E0_AUTO_FAN_PIN to -1 ?

if that is not allready the case (have not checked)

boelle, 2020-02-02:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/stm32/pins_BTT_SKR_MINI_E3.h

so there is no autofan defined so would say there is no bug as the define must have been added by the user

kramer04, 2020-02-02:
ok thanks
so this isn't a bug
i close it

closed 2020-02-02 by kramer04

Talha909, 2020-02-06:
> ok thanks
> so this isn't a bug
> i close it

HI, I have the same problem. Does your fan on automatically when temperature touch 55