MarlinFirmware/Marlin #16764: [FR] MKS Base 1.6 support

DanielMazurkiewicz, 2020-02-02:
Here is the picture of board:

https://cdn11.bigcommerce.com/s-mm9wpa/images/stencil/original/products/1212/5066/MKS_BASE_1.6_3d_printer_controller_01__71237.1506699962.jpg?c=2

Here are all detailed information about board:
https://github.com/makerbase-mks/MKS-BASE/tree/master/hardware

sjasonsmith, 2020-02-02:
Does this board work fine if you use the MKS Base 1.5 motherboard (MKS_BASE_15)?

I can't find much information about changes between these versions, other than some indication that there might not be anything different from a firmware perspective.

DanielMazurkiewicz, 2020-02-02:
> Does this board work fine if you configure your firmware for the MKS Base 1.5 motherboard 

@sjasonsmith I wouldn't risk "just trying it" :-) But if someone have schematics for the 1.5 version (which I couldn't find) then I can do the job and check what are the changes (if any)

yet-another-average-joe, 2020-02-03:
Just use MKS_BASE_15

My Tornado came with a MKS Base 1.6 ; with older Marlin 1.1.x, I've been using  BOARD_MKS_BASE
With Marlin 2.0.x, and for the last 18 months, I used BOARD_MKS_BASE_15 ; I also tested with BOARD_MKS_BASE_14 

Tested with three ABL probes : 3DTouch, BLTouch, and TouchMI ; also with a runout sensor. Tested these probes and sensors on differents ports. And with different hotends (Tevo, V6 and Volcano). Heavily modded printer, with many custom settings + some firmware hacks, RasPi 3B, RasPi 3B+, running OctoPrint and Octolapse.

**Never had an issue.**

Ports # / pinout are the same and they all use Allegro A4982 drivers.

DanielMazurkiewicz, 2020-02-04:
@yet-another-average-joe Have Tevo Tornado too and just used MKS BASE 15 board and screen definitely is not working properly which means somewhere are wrong pin assignments

sjasonsmith, 2020-02-04:
Are using the Tornado example configuration, or creating one from scratch? Presumably whoever created the example would have configured it for the screen, but I haven’t checked in detail.

yet-another-average-joe, 2020-02-04:
> 
> 
> @yet-another-average-joe Have Tevo Tornado too and just used MKS BASE 15 board and screen definitely is not working properly which means somewhere are wrong pin assignments

Sounds weird... Never had a problem with the display.

Tevo Tornado came with at least 4 motherboards (MKS Base 1.4, MKS Base 1.5, MKS Base 1.6 and MKS Gen L) within a few months. I got a 1.6 while some youtubers were still describing the 1.5, and a few weeks later it was the Gen L. Are you sure yours is labeled MKS Base 1.6 ? The original firmwares, 2 years ago, came with Tornado 1.0 and Tornado 1.1, respectlively Marlin 1.1.4 and 1.1.5 AFAIK , and BOARD_MKS_BASE is defined in Configuration.h. Could your board be a MKS Gen L ? The MKS Base 1.4 and Gen L have Polulu formfactor drivers. Base 1.5 and 1.6 have chips soldered on the PCB.

yet-another-average-joe, 2020-02-04:
Just thinking of this : did you define YOUR display in Configuration.h ? The default display if an older one.

DanielMazurkiewicz, 2020-02-04:
@sjasonsmith Yes, used TT example config from Config repo, I can see something on screen but it is noised with some artifacst. Will check it anyway, thanks!

@yet-another-average-joe Thanks for the details and for configs, will check it. My tevo is stock one, no mods on it, and it came with base 1.6.

Just made it, so will drop it here, as it might be useful anyway.

(BTW. I see from schematics that I can adjust steps divider programmatically, is there in Marlin somewhere options for it?)

PIN NAME | PIN NUMBER |   | Arduino „DIGITAL” | Arduino „ANALOG” |   | Comment
-- | -- | -- | -- | -- | -- | --
LED | 26 |   | 13 |   |   |  
  |   |   |   |   |   |  
X_STEP | 97 |   | 54 | 0 |   |  
X_DIR | 96 |   | 55 | 1 |   |  
X_EN | 50 |   | 38 |   |   |  
X_MIN | 7 |   | 3 |   |   |  
X_MAX | 6 |   | 2 |   |   |  
  |   |   |   |   |   |  
X_MS1 | 5 |   | 5 |   |   |  
X_MS2 | 15 |   | 6 |   |   |  
  |   |   |   |   |   |  
Y_STEP | 91 |   | 60 | 6 |   |  
Y_DIR | 90 |   | 61 | 7 |   |  
Y_EN | 95 |   | 56 | 2 |   |  
Y_MIN | 64 |   | 14 |   |   |  
Y_MAX | 63 |   | 15 |   |   |  
  |   |   |   |   |   |  
Y_MS1 | 92 |   | 59 | 5 |   |  
Y_MS2 | 93 |   | 58 | 4 |   |  
  |   |   |   |   |   |  
Z_STEP | 38 |   | 46 |   |   |  
Z_DIR | 36 |   | 48 |   |   |  
Z_EN | 89 |   | 62 | 8 |   |  
Z_MIN | 46 |   | 18 |   |   |  
Z_MAX | 45 |   | 19 |   |   |  
  |   |   |   |   |   |  
Z_MS1 | 78 |   | 22 |   |   |  
Z_MS2 | 70 |   | 39 |   |   |  
  |   |   |   |   |   |  
E0_STEP | 74 |   | 26 |   |   |  
E0_DIR | 72 |   | 28 |   |   |  
E0_EN | 76 |   | 24 |   |   |  
  |   |   |   |   |   |  
E0_MS2 | 87 |   | 64 | 10 |   |  
E0_MS1 | 88 |   | 63 | 9 |   |  
  |   |   |   |   |   |  
E1_STEP | 54 |   | 36 |   |   |  
E1_DIR | 56 |   | 34 |   |   |  
E1_EN | 60 |   | 30 |   |   |  
  |   |   |   |   |   |  
E1_MS2 | 1 |   | 4 |   |   |  
E1_MS1 | 94 |   | 57 | 3 |   |  
  |   |   |   |   |   |  
HEATER_FAN | 18 |   | 9 |   |   |  
HEATER_E0 | 23 |   | 10 |   |   |  
HEATER_E1 | 16 |   | 7 |   |   |  
HEATER_BED | 17 |   | 8 |   |   |  
  |   |   |   |   |   |  
A13_THERM2 | 84 |   | 67 | 13 |   |  
A15_THERM0 | 82 |   | 69 | 15 |   |  
A15_THERM1 | 83 |   | 68 | 14 |   |  
  |   |   |   |   |   |  
LCD_RS | 13 |   | 16 |   |   | EXP1 connector
LCD_EN | 12 |   | 17 |   |   | EXP1 connector
LCD_D4 | 77 |   | 23 |   |   | EXP1 connector
LCD_D5 | 75 |   | 25 |   |   | EXP1 connector
LCD_D6 | 73 |   | 27 |   |   | EXP1 connector
LCD_D7 | 71 |   | 29 |   |   | EXP1 connector
  |   |   |   |   |   |  
BEEPER | 53 |   | 37 |   |   | EXP1 connector
  |   |   |   |   |   |  
BTN_EN1 | 59 |   | 31 |   |   |  
BTN_EN2 | 57 |   | 33 |   |   |  
BTN_ENC | 55 |   | 35 |   |   | EXP1 connector
  |   |   |   |   |   |  
SD_DET | 35 |   | 49 |   |   | EXP2 connector
SD_SCK | 20 |   | 52 |   |   | EXP2 connector / ICSP connector
SD_MISO | 22 |   | 50 |   |   | EXP2 connector / ICSP connector
SD_MOSI | 21 |   | 51 |   |   | EXP2 connector / ICSP connector
SD_SS | 19 |   | 53 |   |   | EXP2 connector
KILL | 51 |   | 41 |   |   | EXP2 connector / pulled up to 5V
  |   |   |   |   |   |  
PE1_D1 | 3 |   | 1 |   |   |  
PE0_D0 | 2 |   | 0 |   |   |  
  |   |   |   |   |   |  
ADC12 | 85 |   | 66 | 12 |   | XS3 Connector
ADC11 | 86 |   | 65 | 11 |   | XS3 Connector
  |   |   |   |   |   |  
SERVOS_D12 | 25 |   | 12 |   |   | Servos connector
SERVOS_D11 | 24 |   | 11 |   |   | Servos connector
  |   |   |   |   |   |  
USB_SS | 43 |   | 21 |   |   |  
USB_BZ | 44 |   | 20 |   |   |

yet-another-average-joe, 2020-02-04:
Yes you can adjust steps/mm (Configuration.h) and microstepping (Configuration_adv.h).

For the Tevo Tornado, the display should be defined "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER"

You could install WinMerge (or similar if you're not a Windows guy), and compare original config files with yours and with vanila ones. If you compare yours with mine, do NOT take geometry, ABL and PID parameters into account !

DanielMazurkiewicz, 2020-02-04:
@yet-another-average-joe I'm using meld and I'm not Windows user either. And meld was first thing I did to see config diffs, there are a lot of differences, but I didn't find anything in particular that "looked important" that is why I assumed that there must be some pin misconfiguration. Unfortunately pin configurations are not easy to follow in Marlin as pins are defined in multiple places, but I'll try to dig it in as well as differences in config files.

DanielMazurkiewicz, 2020-02-04:
Made pinout for V1.6 for Marlin based on schematics, but have no idea if I'm missing something here, is it enough?
```

#define LED_PIN 13
 
#define X_STEP_PIN 54
#define X_DIR_PIN 55
#define X_ENABLE 38
#define X_MIN_PIN 3
#define X_MAX_PIN 2
 
#define X_MS1_PIN 5
#define X_MS2_PIN 6
 
#define Y_STEP_PIN 60
#define Y_DIR_PIN 61
#define Y_ENABLE_PIN 56
#define Y_MIN_PIN 14
#define Y_MAX_PIN 15
 
#define Y_MS1_PIN 59
#define Y_MS2_PIN 58
 
#define Z_STEP_PIN 46
#define Z_DIR_PIN 48
#define Z_ENABLE_PIN 62
#define Z_MIN_PIN 18
#define Z_MAX_PIN 19
 
#define Z_MS1_PIN 22
#define Z_MS2_PIN 39
 
#define E0_STEP_PIN 26
#define E0_DIR_PIN 28
#define E0_ENABLE_PIN 24
 
#define E0_MS2_PIN 64
#define E0_MS1_PIN 63
 
#define E1_STEP_PIN 36
#define E1_DIR_PIN 34
#define E1_ENABLE_PIN 30
 
#define E1_MS2_PIN 4
#define E1_MS1_PIN 57
 
#define FAN_PIN 9
#define HEATER_0_PIN 10
#define HEATER_1_PIN 7
#define HEATER_BED_PIN 8
 
#define TEMP_2_PIN 13
#define TEMP_0_PIN 15
#define TEMP_1_PIN 14
 
#define LCD_PINS_RS 16
#define LCD_PINS_ENABLE 17
#define LCD_PINS_D4 23
#define LCD_PINS_D5 25
#define LCD_PINS_D6 27
#define LCD_PINS_D7 29
 
#define BEEPER_PIN 37
 
#define BTN_EN1 31
#define BTN_EN2 33
#define BTN_ENC 35
 
#define SD_DETECT 49
#define SCK 52
#define MISO 50
#define MOSI 51
#define SDSS 53
#define KILL_PIN 41
 
 
 
 
 
 
 
#define SERVO1_PIN 12
#define SERVO0_PIN 11
```

thisiskeithb, 2020-02-05:
Take a look at #16783. Marlin will report the correct board name if you define `MKS_BASE_15` or `MKS_BASE_16`.

DanielMazurkiewicz, 2020-02-05:
@thisiskeithb Thank you for adding the board! I saw you merged it already, but added a comment to your PR, check #16783

DanielMazurkiewicz, 2020-02-05:
@sjasonsmith 
@yet-another-average-joe 
@thisiskeithb 
Did in depth investigation on configs, pins, spent couple of hours on it, but finally solved issue with screen. The problem wasn't in pinouts but in data timings. Details here: https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller#Issues

Added option to LCD configuration and that fixes issue:
```
//=============================================================================
//=======================   LCD / Controller Selection  =======================
//=========================      (Graphical LCDs)      ========================
//=============================================================================

//
// CONTROLLER TYPE: Graphical 128x64 (DOGM)
//
// IMPORTANT: The U8glib library is required for Graphical Display!
//            https://github.com/olikraus/U8glib_Arduino
//

//
// RepRapDiscount FULL GRAPHIC Smart Controller
// http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
//
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

// Enable this option if you have issues with screen image corruption
//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER_ADJUST_TIMINGS
#if ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER_ADJUST_TIMINGS)
    #define ST7920_DELAY_1 DELAY_NS(0)
    #define ST7920_DELAY_2 DELAY_NS(0)
    #define ST7920_DELAY_3 DELAY_NS(127)
#endif
```

Would be probably nice to have it in default config.

yet-another-average-joe, 2020-02-05:
Looking at the Tevo Tornado example files, we see that these lines were added to v2 (MKS Base Gen L) Configuration.h :

```
#define ST7920_DELAY_1 DELAY_NS(150)
#define ST7920_DELAY_2 DELAY_NS(150)
#define ST7920_DELAY_3 DELAY_NS(150)

```
But nothing in Tornado v1

closed 2020-02-09 by thinkyhead