MarlinFirmware/Marlin #16766: [BUG] MAX31865 on SKR Pro V1.1 on bugfix 2.0.x

PM2189, 2020-02-02:
Marlin will not boot with firmware with MAX31865 with a PT100 on Extruder 0.

Board: BTT SKR Pro V1.1
Drivers: TMC5160 on X,Y,Z,E0
I²C EEPROM module
 
Extruder Temperature by PT100 RTD
Bed Temperature by 100KΩ Thermistor
Chamber Temperature by 10KΩ Thermistor

MAX31865 MISO, MOSI, and SCK pins were set to the same as TMC5160's (PC11, PC12, PC10). CK was set to PF7.

MAX31865 board was tested and found functional.

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4145710/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4145711/Configuration_adv.txt)
[pins_BTT_SKR_PRO_V1_1.txt](https://github.com/MarlinFirmware/Marlin/files/4145707/pins_BTT_SKR_PRO_V1_1.txt)

afkfurby, 2020-02-10:
Is it possible that the max31865 generally don't work on stm32 boards with marlin?

bdelia, 2020-04-14:
Not sure this will help, but I have seen that when I set my TEMP_SENSOR_0 = 20 so that I can use a PT100 amplifier I get similar results of the FW compiling without issues but the board not posting or getting recognized by Windows.  If I then keep all configurations in place by assign TEMP_SENSOR_0 = 1 everything works.  I am beginning to wondering if there is a defect relating to the PT100 amp board selection for the TEMP_SENSORs

wormley, 2020-06-03:
This doesn't look like an SPI problem. 

There are a number of things that I don't like in the MAX31865 code and I'm working to make changes and ultimately a PR, but the first one that appears to cause this bug is in the function: 
updateTemperaturesFromRawValues() where it has:
            #if ENABLED(MAX6675_IS_MAX31865)
              max31865.temperature(100, 400)  // 100 ohms = PT100 resistance. 400 ohms = calibration resistor

This function doesn't want the current temperature in all cases, one time it's used is to try and figure out what the max and minimum raw values are if you set limits.(at the define for _TEMP_MIN_E ) In this case the code gets into an infinite loop since the return value never changes. The fix for that issue is to simply replace the function call with return(raw); as the raw value from the READ call is just the temp.

Of course, now I'm hitting a MINTEMP error so I need to do more debugging and see if the sensor is being read correctly. But at least it's not hanging without an error.

AnHardt, 2020-06-03:
> #if ENABLED(MAX6675_IS_MAX31865)
max31865.temperature(100, 400) // 100 ohms = PT100 resistance. 400 ohms = calibration resistor

Why the hell a hard coded 400 Ohm for the reference resistor?
The [Adafruit modules](https://www.adafruit.com/product/3328) and all known Chinese clons  do have a 430 +-0.1% Ohm resistor.

sjasonsmith, 2020-06-03:
We downgraded the dependency from master to 1.1 to fix compilation broken in 1.2. It is possible that there are important fixes between those two versions, and we need to depend on a specific commit rather than a released version. Just something to consider.

wormley, 2020-06-03:
It all looks fine with 1.1 the change from 1.1 to 1.2 appears to just be moving the SPI code from internal to the BusIO library.

I've now enabled TMC on SPI on the same pins as the MAX31865 and nothing has crashed yet and the temperatures look sane. My branch is nowhere near ready to merge, mostly because I broke MAX6675 in the process, but can be viewed here for a diff:
https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...wormley:local

boelle, 2020-06-21:
@PM2189 still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle