MarlinFirmware/Marlin #16767: [BUG] latest bugfix compile error for due

maukcc, 2020-02-03:
### Bug Description

When compiling for due we get an error:

> In file included from sketch/src/MarlinCore.cpp:39:0:
> sketch/src/module/probe.h:69:40: error: could not convert '0' from 'int' to 'XYZval<float>::<anonymous union>'
>      static constexpr xyz_pos_t offset{0};
>                                         ^
> sketch/src/module/probe.h:70:42: error: could not convert '0' from 'int' to 'XYval<float>::<anonymous union>'
>      static constexpr xy_pos_t offset_xy{0};
>                                           ^
> Multiple libraries were found for "Wire.h"
>  Used: /home/maukcc/.arduino15/packages/arduino/hardware/sam/1.6.12/libraries/Wire
> exit status 1
> Error compiling for board Arduino Due (Native USB Port).
> 

This does not happen when compiling for MEGA

ellensp, 2020-02-03:
Please provide failing config files.
I compiled for BOARD_RADDS and BOARD_RAMPS_DUO_EFB, both compiled no issues

maukcc, 2020-02-04:
Here you go
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4151959/Configuration.zip)

thinkyhead, 2020-02-04:
Using your configs I get the result
```
DUE                           SUCCESS   00:00:54.954
```
Re-download a fresh copy of Marlin and try again.

closed 2020-02-04 by thinkyhead

maukcc, 2020-02-04:
With todays fresh bugfix-2 download and only changed the 2 config files, we still get the same error.
Arduino IDE 1.8.10 with arduino due native port selected. (programming port gives the same results)

thinkyhead, 2020-02-04:
I used your configs and they compiled very happily, so…… it probably ain't on our end. 🤷‍♂

ellensp, 2020-02-04:
Compiles on platformio without issues, but this does fail on arduino ide 1.8.9 and 1.8.10 (both Linux in my case)

thinkyhead, 2020-02-04:
How annoying! Okay, see if this makes it happy:
```
static constexpr xyz_pos_t offset { 0, 0, 0 };
static constexpr xy_pos_t offset_xy { 0, 0 };
```
…or this…
```
static constexpr xyz_pos_t offset = { 0, 0, 0 };
static constexpr xy_pos_t offset_xy = { 0, 0 };
```
…or even this…
```
static constexpr xyz_pos_t offset;
static constexpr xy_pos_t offset_xy;
```
One or all of them should satisfy that eccentric IDE.

ellensp, 2020-02-04:
Nope, Nope and nope.  not one works.

thinkyhead, 2020-02-04:
Damn but Arduino takes forever to compile. If you value your sanity, people, upgrade to VSCode.

thinkyhead, 2020-02-04:
FFS Arduino!
```
.../Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags:
  bad CPU type in executable
```

ellensp, 2020-02-04:
I'm waiting for the day you say marlin no longer supports Arduino IDE..  it would solve a lot of problems.

thinkyhead, 2020-02-04:
Seems I had 1.8.9 installed on this laptop. Turning on app folder syncing now…

thinkyhead, 2020-02-04:
Well, let's get crazy then…
```cpp
static constexpr xyz_pos_t offset = xyz_pos_t({ 0, 0, 0 });
static constexpr xy_pos_t offset_xy = xyz_pos_t({ 0, 0 });
```

ellensp, 2020-02-04:
crazy works, after fixing your typo
    
```
static constexpr xyz_pos_t offset = xyz_pos_t({ 0, 0, 0 });
static constexpr xy_pos_t offset_xy = xy_pos_t({ 0, 0 });

```

maukcc, 2020-02-04:
Thanks for going the extra meter.

thinkyhead, 2020-02-04:
Odd that Arduino's compiler isn't bothered by the original syntax in other places. I'll merge a patch shortly.

commit 081ef46e (2020-02-04): Fix Arduino IDE compile for DUE

Fixes #16767