MarlinFirmware/Marlin #16768: [BUG] No EEPROM suddenly!

karabas2011, 2020-02-03:
### Bug Description

 MARLIN 2.0.x snapshot 30.01.2020
SKR 1.1pro X TMC2208 UART

M500 returns "no EEPROM"
But EEPROM enabled in config
M115 returns EEPROM:1

### My Configurations
Cartesian IDEX extruders 

**Required:** Please in

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4147072/Marlin.zip)

clude a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

italocjs, 2020-02-03:
i'm not sure if this will help, but i had some really weird issues caused by corrupted eeprom data, i fixed it by running a simple arduino program to set all eeprom bits to zero, and then burning the firmware again.  (there is a sketch inside arduino examples if i'm not mistaken)

karabas2011, 2020-02-03:
It works for arduino, but not for STM32F4
If I use older version all works. But I need fixes for manual mesh leveling

italocjs, 2020-02-03:
I see, it might be something in this version, unfortunately i dont have this board, so i cant help much on looking out this issue.

spattinson, 2020-02-03:
I had this earlier today with my skr pro. I defaulted my settings then couldn't save any new ones.  a reformat, a reboot fixed plus a few M501 M502 M500 then M503 to view them. For now I am sticking to 8.3 filenames for gcode files idk what the cause is. It may be worth doing a filesystem check (scan  for errors or fscl on  linux)

karabas2011, 2020-02-04:
Incorrect. EEPROM is NOT sdcard on this board. Older versions works. You dont need sdcard at al after flashing.

thinkyhead, 2020-02-04:
Check out the pins file.
```cpp
// Use one of these or SDCard-based Emulation will be used
//#define SRAM_EEPROM_EMULATION   // Use BackSRAM-based EEPROM emulation
//#define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation
```

thinkyhead, 2020-02-04:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-02-04 by thinkyhead