MarlinFirmware/Marlin #16770: static assertion failed: RIGHT_PROBE_BED_POSITION is outside the probe region.[BUG] (short description)

ElTel15, 2020-02-03:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ElTel15, 2020-02-03:
I think that this is a bug in 1.1.9(Bugfix).  I have done extensive research and have seen this crop up regularly especially when using and configuring a BL touch.  
I configured the software using microswitch endstops on all 3 axis of my self built reprap cartesian printer and the printer responded to all commands correctly.  However I then installed a BT Touch sensor and enabled ABL Bilinear which I believe that I configured accurately. I  set the offset of the nozzle on the X axis (-ve) in order for it to centre on the bed and set the sensor offset (again -ve).  I have set the bed size slightly smaller than the actual bed and the edge non probing area to 10 mm. 

When compiling I get the sanity check: 

**static assertion failed: RIGHT_PROBE_BED_POSITION is outside the probe region.**

I have tried changing the bed size (both increased and decrease) as well a altering the probe offset to zero and a +ve figure but still get the same sanity check.  I have not edited the Configuration_adv.h file

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4147213/Configurations.zip)

**To reproduce** 
1. unzip the files and then 
2. copy and paste the configuration files into Marlin 1.1.9 (Bugfix) in Arduino IDE 1.8.10 to replace the original file Configuration.h and and Configuration_adv.h and then,
3. compile using Arduino IDE

I expected to be able to compile the firmware.

It did not compile and returned the above sanity check.

I have consulted similar reports on GitHub and have not seen a successful answer to the problem only work arounds plus I am not a skilled enough programmer to analyse the problem.

ElTel15, 2020-02-03:
P.S. I should have said that I have searched various forums including RepRap and have seen the problem discussed but no solution so far.  I have opened a topic on the RepRap forum hoping to get an insight into the problem (Bug?).

TerryD

boelle, 2020-02-03:
there is no Marlin 1.1.9 (Bugfix)

there are only 1.1.x bugfix

2.0.x bugfix is the only one actived work on so please try that first

you can use a tool like winmerge to compare config files and copy over settings, you cant just reuse old config files

ellensp, 2020-02-03:
Using your config files and enabling BLTOUCH and AUTO_BED_LEVELING_BILINEAR compiles on bugfix-1.1.x without any RIGHT_PROBE_BED_POSITION issues. I suspect you have found a fug that has already been fixed.

ElTel15, 2020-02-06:
Sorry but I attached the wrong config zip file am attepting to use Marlin 2'0'x but am having dificulty with an error1: Error compiling for board Arduino Mega or Mega 2560. and I'm stuch here as I,ve had no help from any of the normal sources but I'll probably solve it or throw the printer in teh bin a I've been at it for weeks now with no success but 8 years ago I had no problem with basic configuration and using machine.  Sorry again just a mild personal rant.

ellensp, 2020-02-06:
and yet you fail to provide the correct config file...   that trigger this error.

ElTel15, 2020-02-06:
I did not need to as the latter post was a remark rather that=n a request fior help.  I have now solved the problem asn it was not a configuraation error as such - I knew that was correct, it was an Arduino error and I used Platformio to check my edits and ompile the firmwaare.  So thank you for taking the time to read and respond to my comments, it was most helpful in teh ed.

Terry D

closed 2020-02-06 by ElTel15