MarlinFirmware/Marlin #16778: stm32f103rb: repalce HAL_STM32F1 to HAL_STM32

whoim2, 2020-02-04:
I'm trying to fit marlin 2.0 in stm32f103rbt6 without display and sd card support, based on the config for kossel. I have a little lack of processor memory. I heard that when compiling through HAL_STM32 instead of HAL_STM32F1, the firmware is less. Can anyone advise me on a config for platformio.ini for this?
Thanks

boelle, 2020-02-04:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-02-04 by boelle

thinkyhead, 2020-02-04:
@boelle — If you want the actual Markdown for that canned reply, it is in the message above. You can use the "Edit" option to view and copy it, then paste that into your notes app. The Markdown is more readable and demonstrates that we care about the small details.