MarlinFirmware/Marlin #16779: [BUG] IDEX mode X2 incorrect steps.

karabas2011, 2020-02-04:
### Bug Description

X2 stepper goes 2x of X.  It looks like  it set 1/8 microstep insteed of 1/16 
M122 does NOT show X2 channel at all

### My Configurations
SKR 1.1 pro TMC2208 UART
cartesian IDEX  autopark mode
MARLIN 2.0.x from 30.01.2020

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

Setup, switch to T1
every G1 X10 will move X2 carriage to 20mm

TheNeskik, 2020-02-04:
In the configuration_adv file, you did not activate Dual_X_Carriage or Dual_X_Stepper, is this normal?

thinkyhead, 2020-02-04:
Agreed. You're not doing IDEX without enabling `DUAL_X_CARRIAGE`, chief.

boelle, 2020-02-04:
could we not move this support Q to discord instead?

karabas2011, 2020-02-04:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4152326/Marlin.zip)

thinkyhead, 2020-02-04:
You're certain all the jumpers on the board are consistent?

thinkyhead, 2020-02-04:
Not that it would matter for this issue, but `M502`, `M500` to Reset EEPROM can never hurt.

thinkyhead, 2020-02-04:
I'm thinking this could potentially be a hardware issue. Your configs look fine to me.
Are those 1.9° steppers with 12-tooth pulleys or 0.8° steppers with 24-tooth pulleys?

karabas2011, 2020-02-04:
1.9 steppers with 12mm/turn leadscrews.
The probles is only on newer snapshot
Today I dl current snapshot - the same.
TMC lib is changed recently. I think some forgotten settings.
Where is step settings for X2?

karabas2011, 2020-02-04:
> Not that it would matter for this issue, but `M502`, `M500` to Reset EEPROM can never hurt.

My friend got the same problem. He pointed me on it and I confirmed.

karabas2011, 2020-02-04:
The same printer - snapshot from 30.11.2019
Connecting...
Printer is now online.
>>> M122
SENDING:M122
Testing X connection... OK
Testing X2 connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK

So we see that last snapshot does init driver in standalone mode but not in UART mode

boelle, 2020-02-04:
> 12mm/turn leadscrews.

so no belt and pulley? @thinkyhead wanted to know about those

karabas2011, 2020-02-04:
> > 12mm/turn leadscrews.
> 
> so no belt and pulley? @thinkyhead wanted to know about those

exactly.

karabas2011, 2020-02-04:
thе only difference is  snapshot date.

boelle, 2020-02-04:
just curious, is it a self build printer? first time i see a printer without belts for X and Y

karabas2011, 2020-02-04:
We are developed such printers. To be sure that they are good or not.

InsanityAutomation, 2020-02-04:
@boelle its actually fairly common when you start to hit the industrial range.

TMC drivers will hang if the step rate gets too fast until enable is dropped, or sometimes even a power cycle. Could try increasing the pulse timings a bit for that problem.

Only issue I have chasing it is my idex machines are all standalone drivers, not uart. Ill do a little poking.

karabas2011, 2020-03-04:
Does anybody able to replicate the problem?

InsanityAutomation, 2020-03-04:
Sorry, day job and wrapping up mrrf projects have eaten all my time. I'll get to it at some point... Hopefully sooner than later...

thinkyhead, 2020-03-06:
> Does anybody able to replicate the problem? or any possibility to set microsteps to X2?

Checking the code it looks like all the drivers are initialized in the same way and at the same time.

Try sending `M502` followed by `M122` a few times and see if anything changes.

thinkyhead, 2020-03-06:
Here are your configs updated for the latest Marlin code: [config-204.zip](https://github.com/MarlinFirmware/Marlin/files/4301049/config-204.zip). Give `bugfix-2.0.x` a try and see if the same issue appears.

karabas2011, 2020-03-10:
>>> M906
SENDING:M906
X driver current: 800
X2 driver current: 800
Y driver current: 800
Z driver current: 800
E driver current: 800
E1 driver current: 800

All works!!!  Thanx!
snapshot from 05.03.2020

closed 2020-03-11 by thinkyhead