MarlinFirmware/Marlin #16780: [FR] New Gcode (M74?) for "Start of new print"

wavexx, 2020-02-04:
### Description

It seems that there's no easy way for a usb print host to signal to the printer that it's about to start a new print job.

Sure we have ways to set the print status, but I don't see anything the FW could use to perform some internal bookkeeping when a new print starts.

Documentation of M31 shows that if PRINTJOB_TIMER_AUTOSTART is enabled, M109/M190 will do something similar (reset job timers), but it seems more of a hack than anything else.

I've been working for a while on Prusa's fork of Marlin 1.1.x and I was looking for a way to reset print stats properly at the beginning of a print (https://github.com/prusa3d/Prusa-Firmware/issues/676) when a usb host is used.

In similar circumstances I actually abused G29 instead of M109 since it's often performed once per print, but I'd love to do something clean and that's futureproof.

What about introducing a new gcode that can be inserted by the host? Looking at reprap and marlin docs, it seems that M74 would be unused.

### Feature Workflow

The new gcode could be a perfect substitute for PRINTJOB_TIMER_AUTOSTART.
A good description would be "Signal start of new print job", and could be placed as the very first instruction a usb host sends.

Since unknown gcodes are effectively ignored, doing nothing at first would still be perfectly acceptable as long as we agree on what this would do.

This new gcode would be a perfect place to reset job times and last print statistics.

Slicers can simply insert this in the start gcode sequence. When printing via sd this is not needed, but it would improve the behavior with usb hosts that don't support this yet.

A print host could later do this on it's own.

iosonopersia, 2020-02-04:
MK4duo already has this type of command. In its specific case it's the M530 S1 command (for "start of the print") and M530 S0 (for "end of the print"). I suggest not to cause too much trouble for the host developers by using the same code for the same operation even across different firmwares.
@MagoKimbra

thinkyhead, 2020-02-04:
At this time we support `M75`, `M76`, `M77` and hosts generally know the dialect. `M530 S1` would be an alias for `M75` and `M530 S0` would be an alias for `M77`.

wavexx, 2020-02-04:
> MK4duo already has this type of command. In its specific case it's the M530 S1 command (for "start of the print") and M530 S0 (for "end of the print").

Could you point to a some description? What I read on the lift of supported gcodes is a bit vague to me.

> I suggest not to cause too much trouble for the host developers by using the same code for the same operation even across different firmwares.

So you would suggest to _not_ even attempt this?

thinkyhead, 2020-02-04:
The G-codes are simply invented by developers as we go along and are adopted on an _ad hoc_ basis. When there is a compelling demand we will add a new feature enabling some new G-code, but it won't be enabled by default, because that's how we roll out new features. So, users will need to enable it and hosts will need to choose it over `M75`, `M76`, `M77`. So, if the code is not in common usage and filling in for some missing function, we can take it or leave it.

wavexx, 2020-02-04:
I'd actually re-implement M75/77 in prusa's fw then, they seem fitting and reasonably easy to match in behavior too.

MagoKimbra, 2020-02-04:
No sorry, but it's not like that. Repetier Host and FW were the first to use M530 as a start command and not only, the file name, the number of layers and other information are also sent to the fw. MK4duo adopted this gcode long before Marlin made M75 so eventually M75 is an alias of M530 S1.
For me it would be right that the commands that have already been adopted were not changed with others otherwise we make it worse.

thinkyhead, 2020-02-07:
No one submitted a `M530` feature to the Marlin project, but someone submitted `M75`-`M77` codes to control the print job timer, so that's all we've got today. While we are waiting for someone to implement `M530`, the print job timer control codes are a reasonable substitute in Marlin.

From what I gather, the `M530` code was created as a way for Repetier Host and Repetier Firmware to work together and they had their own ideas about that. Marlin hasn't got any use for the name of the file being printed by the host, so the filename sent by `M530` is currently not needed, so whoever decides to implement the code can leave that out.

wavexx, 2020-02-07:
I think this describes everything I needed to know, thank to all.
I don't want to keep this open as an "issue", so I will close this.

Thanks again

closed 2020-02-07 by wavexx