MarlinFirmware/Marlin #16782: Errors Trying to Build: error: missing binary operator before token "("

dliving2010, 2020-02-05:
I am trying to run another build on my project, which I built and uploaded before without any issues.  Originally I did this in VS Code, but because it kept getting the errors, I tried Atom, but I'm getting the same errors there as well.  What is strange is that I haven't changed anything in adv.h, .h or platformio.ini, yet I am now getting the following errors whenever I try to build:

PackageManager: Installing tool-pyocd @ ~0.801.0
Downloading...
Unpacking...
tool-pyocd @ 0.801.0 has been successfully installed!
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; framework: arduino; board: nxp_lpc1768)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option

CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.2.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.5.2
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in debug mode
 
Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk
 
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
In file included from c:\users\dan\desktop\temp\marlin\src\inc\MarlinConfigPre.h:54,
                 from c:\users\dan\desktop\temp\marlin\src\inc\marlinconfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
c:\users\dan\desktop\temp\marlin\src\core\drivers.h:68:51: error: missing binary operator before token "("
#define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^

c:\users\dan\desktop\temp\marlin\src\core\drivers.h:91:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~
c:\users\dan\desktop\temp\marlin\configuration_adv.h:2058:7: note: in expansion of macro 'AXIS_IS_TMC'
   #if AXIS_IS_TMC(Z4)
       ^~~~~~~~~~~
c:\users\dan\desktop\temp\marlin\src\core\drivers.h:68:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
c:\users\dan\desktop\temp\marlin\src\core\drivers.h:91:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~
c:\users\dan\desktop\temp\marlin\configuration_adv.h:2108:7: note: in expansion of macro 'AXIS_IS_TMC'
   #if AXIS_IS_TMC(E6)
       ^~~~~~~~~~~
c:\users\dan\desktop\temp\marlin\src\core\drivers.h:68:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
c:\users\dan\desktop\temp\marlin\src\core\drivers.h:91:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~
c:\users\dan\desktop\temp\marlin\configuration_adv.h:2115:7: note: in expansion of macro 'AXIS_IS_TMC'
   #if AXIS_IS_TMC(E7)
       ^~~~~~~~~~~
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\fast_pwm.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:54,
[... 123 lines cut ...]
STM32F103RC_bigtree_512K_NOUSB  IGNORED
STM32F103RE_bigtree             IGNORED
STM32F103RE_bigtree_NOUSB       IGNORED
STM32F4                         IGNORED
STM32F7                         IGNORED
ARMED                           IGNORED
STM32F103VE_longer              IGNORED
mks_robin                       IGNORED
mks_robin_lite                  IGNORED
mks_robin_mini                  IGNORED
mks_robin_nano                  IGNORED
jgaurora_a5s_a1                 IGNORED
STM32F407VE_black               IGNORED
BIGTREE_SKR_PRO                 IGNORED
BIGTREE_BTT002                  IGNORED
teensy31                        IGNORED
STM32F103CB_malyan              IGNORED
chitu_f103                      IGNORED
teensy35                        IGNORED
esp32                           IGNORED
fysetc_f6_13                    IGNORED
linux_native                    IGNORED
SAMD51_grandcentral_m4          IGNORED
include_tree                    IGNORED
==================== 1 failed, 0 succeeded in 00:00:02.424 ====================

ellensp, 2020-02-05:
Please attach configuration files as requested

boelle, 2020-02-05:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

dliving2010, 2020-02-05:
Last night I decided to just delete the project and download a fresh copy of Marlin.  After reconfiguring, I was able to build and upload without issue.

boelle, 2020-02-05:
if so could you please hit the close button below?

closed 2020-02-05 by dliving2010