MarlinFirmware/Marlin #16784: [BUG] RepRapDiscount 12864 speaker behaves weird on SKR v1.4

VanessaE, 2020-02-05:
### Bug Description

I have a brand new RepRapDiscount 12864 display, and a brand new SKR v1.4.  Marlin is configured to assume the sound device is a speaker (not a buzzer).

Playing tones through the speaker just results in "garbled" or "warbling" sounds instead of pure tones of the requested frequency.  Listening closely, there are two tones present:  my requested pitch, and a second, high-pitch whine of a steady frequency.

I played around with different pitches, and they all sounded like they are probably correct frequency, but with the high-pitch "overlaid" on top.  Go all the way down to a requested pitch of 2 Hz and you basically get a pulsing high pitch tone (instead what should just sound like clicking).

Fearing a hardware fault, I put my 'scope on the speaker output (that is, the collector side of the drive transistor) on the RRD board.

According to the scope, each tone I ask for produces a square wave of the requested pitch and duration, with an amplitude of about 1v.  Overlaid on the highs of that square wave, there's my high pitch whine, with an amplitude of a few tenths of a volt.  Something like this:
[image]
(sorry I didn't get a chance to take a photo of the scope's screen before putting everything away).

I have two RRD 12864's, both of which work just fine on my SKR v1.1.  It's only on this SKR v1.4 that they misbehave.  I also tried a second set of ribbon cables.

One of them had the resistor "hack"[1] when I started trying to get this working, the other did not (though now it does - a failed last-ditch effort to make it work).

Both 12864's work display Marlin's imagery as intended, and their knobs and buttons work, but I have not yet tested the SD card on the SKR v1.4.

### My Configurations

<details><summary>Configuration.h</summary>
```

```
</details>

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

AnHardt, 2020-02-05:
I guess it's because the "sound devices" on the RepRapDiscount Fullgraphic Smart Controller (128x64) as well as on the RepRapDiscount Smart Controller (20x4) are, by our definition, a "buzzer", producing it's own, manufacturer determined, frequency.

Try to configure as a "buzzer" instead of a "speaker". Is the produced frequency the same as the one you observe as the "high-pitch whine"?

If that's the case, your observation on the 1.4 is the expected one, not that on the 1.1.

VanessaE, 2020-02-05:
Indeed, when so-configured, it produces a steady, clean tone of the exact same frequency.

But why does it work properly as a speaker on the SKR v1.1?

AnHardt, 2020-02-05:
I don't know.
PZ1 draws a considerable amount of current - for that the Q1BC817.
Depending on the signal level of the pin, the coupling to VCC and or GND (pull-up/down) the power of the VCC powersupply or the conductors diameters and decoupling capacitors values and placement - about everything can happen.
The high frequency wave off the pin you see on the  scope could be a variation off VCC or GND coupled in by the 10K pulldown (depending on where the resistor and the scopes probes are exactly placed).

Using "buzzers" as "speakers" always needed some luck. 
Better avoid it. For good reason "speaker" is not the default.

VanessaE, 2020-02-05:
Well I suppose I could replace the buzzer with a real speaker. :smile:

closed 2020-02-05 by VanessaE

VanessaE, 2020-02-06:
In case you're curious:  I did replace the buzzer with a proper piezo speaker (found one in my junk box).  It works fine now.

Sorry for the invalid report.

thinkyhead, 2020-02-07:
> PZ1 draws a considerable amount of current - for that the Q1BC817.

That may explain why playing tones is causing the cooling fan to slow with some boards.

thisiskeithb, 2020-03-28:
> Sorry for the invalid report.

I think it's valid & this issue has cropped up on several BigTreeTech boards, so I put in PR #17320 to undefine `SPEAKER` automatically in the board pins.

VanessaE, 2020-03-28:
No, don't do that...  it's trivial to alter the board to put in a real speaker, and "RepRapDiscount" 12864 displays are not all made by BTT.