MarlinFirmware/Marlin #16785: [FR] G76 reset probe heating height to original position for lower quality probes

tonihoang, 2020-02-05:
### Feature Workflow

Currently when using G76, after each probing the difference to the old value is kept for Z Axis, when returning to the probing point with z Value 0.5.

When using a probe which is more temperature dependant, probe heating positions moves higher and higher, thus is becomes harder or rather takes longer to reach the next temperature for the probe as it moves away from the bed.

An idea to speed up the probing for positive previous z measurements:

1. offsett Z value of
`Line 286:  destination.set(temp_comp.measure_point_x, temp_comp.measure_point_y, 0.5);`
in \Marlin\src\gcode\calibrate\G76_M871.cpp to move z down further by the last `measured_z`, if the previous measurement value was positive.

#### Additional Information

For me the total distance of 2 was not even enough from 30° to 50° C. I got 2.5 at 45° C.
Since I am new to Marlin I still have to understand it all thus I wrote this FR first and then start working on it.

And I receive values above 10 at 50° I guess it might make sense to immediately stop additional measurements instead of waiting for the timeout and then failing with the sanity check in line 139 ff in probe_temp_compensation.cpp.

tompe-proj, 2020-02-13:
@tonihoang 
Using the latest measured z-height for placing the probe should actually not make much of a difference for heating as deltas are in micrometer range. If your first layer is calibrated properly, z-axis=0 should actually be already below trigger height anyway (usually M851 Zxx is set to something like -1mm). However, in line 286 we probably should change the value 0.5 to 0 in order to get the probe even closer to the bed.
`Line 286: destination.set(temp_comp.measure_point_x, temp_comp.measure_point_y, 0.5);`

Here we already had a short discussion about probes not heating properly:
#16718

If the sanity check at line 139 fails for you, you have another problem. Current limitations are a max. delta of 2mm and 0.8mm between two successive probings. If your probe is that inaccurate you better level by hand. Maybe your are losing steps on your z-stepper. Keep in mind that printed values for compensation are in micrometers, so 10 would actually only be 0.01mm.

As of now, a probe should always return to the exact same spot, independent of measured deltas. So if you observe differently you might lose steps while moving back and forth.

tonihoang, 2020-04-11:
so the latest PR fixed the probe going higher and higher for every heating for me, but something different fails now.

I turned on leveling debugging:

some value keeps increasing the sync_plan_position, so the probe tries to move higher and higher:

```
Waiting for probe heating. Bed:110 Probe:30
[...]
>>> do_blocking_move_to X112.00 Y112.00 Z5.00
<<< do_blocking_move_to
>>> Probe::probe_at_point(112.00, 112.00, none, 0, nozzle_relative)
  current_position= X112.00 Y112.00 Z5.00 : 
>>> do_blocking_move_to X112.00 Y112.00 Z5.00
<<< do_blocking_move_to
  current_position= X112.00 Y112.00 Z5.00 : Probe::set_deployed
deploy: 1
Probe::move_z(5.00)
>>> do_blocking_move_to X112.00 Y112.00 Z8.39
<<< do_blocking_move_to
echo:busy: processing
>>> do_blocking_move_to X112.00 Y112.00 Z8.39
<<< do_blocking_move_to
  current_position= X112.00 Y112.00 Z8.39 : >>> Probe::run_z_probe
  current_position= X112.00 Y112.00 Z8.39 : >>> Probe::probe_down_to_z
>>> do_blocking_move_to X112.00 Y112.00 Z1.39
<<< do_blocking_move_to
echo:busy: processing
  current_position= X112.00 Y112.00 Z3.79 : sync_plan_position
  current_position= X112.00 Y112.00 Z3.79 : <<< Probe::probe_down_to_z
```

``  current_position= X112.00 Y112.00 Z3.79 : sync_plan_position``

```
Waiting for probe heating. Bed:110 Probe:35
[...]
>>> do_blocking_move_to X112.00 Y112.00 Z5.00
<<< do_blocking_move_to
>>> Probe::probe_at_point(112.00, 112.00, none, 0, nozzle_relative)
  current_position= X112.00 Y112.00 Z5.00 : 
>>> do_blocking_move_to X112.00 Y112.00 Z5.00
<<< do_blocking_move_to
  current_position= X112.00 Y112.00 Z5.00 : Probe::set_deployed
deploy: 1
  current_position= X112.00 Y112.00 Z5.00 : >>> Probe::run_z_probe
  current_position= X112.00 Y112.00 Z5.00 : >>> Probe::probe_down_to_z
>>> do_blocking_move_to X112.00 Y112.00 Z1.39
<<< do_blocking_move_to
  current_position= X112.00 Y112.00 Z3.91 : sync_plan_position
  current_position= X112.00 Y112.00 Z3.91 : <<< Probe::probe_down_to_z
>>> do_blocking_move_to X112.00 Y112.00 Z8.91
<<< do_blocking_move_to
echo:busy: processing
  current_position= X112.00 Y112.00 Z8.91 : >>> Probe::probe_down_to_z
>>> do_blocking_move_to X112.00 Y112.00 Z1.39
<<< do_blocking_move_to
echo:busy: processing
  current_position= X112.00 Y112.00 Z3.95 : sync_plan_position
  current_position= X112.00 Y112.00 Z3.95 : <<< Probe::probe_down_to_z
```
``  current_position= X112.00 Y112.00 Z3.91 : sync_plan_position``

and at the 4th time it fails cause it tries to move UP

```
Waiting for probe heating. Bed:110 Probe:50
[...]
<<< do_blocking_move_to
>>> Probe::probe_at_point(112.00, 112.00, none, 0, nozzle_relative)
  current_position= X112.00 Y112.00 Z5.00 : 
>>> do_blocking_move_to X112.00 Y112.00 Z5.00
<<< do_blocking_move_to
  current_position= X112.00 Y112.00 Z5.00 : Probe::set_deployed
deploy: 1
  current_position= X112.00 Y112.00 Z5.00 : >>> Probe::run_z_probe
  current_position= X112.00 Y112.00 Z5.00 : >>> Probe::probe_down_to_z
>>> do_blocking_move_to X112.00 Y112.00 Z1.39
<<< do_blocking_move_to
  current_position= X112.00 Y112.00 Z5.61 : sync_plan_position
  current_position= X112.00 Y112.00 Z5.61 : <<< Probe::probe_down_to_z
SLOW Probe fail!
  current_position= X112.00 Y112.00 Z5.61 : <<< run_z_probe
  current_position= X112.00 Y112.00 Z5.61 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to X112.00 Y112.00 Z5.61
<<< do_blocking_move_to
//action:notification Probing Failed
Error:Probing Failed
<<< Probe::probe_at_point
!Received NAN. Aborting.
```

``  current_position= X112.00 Y112.00 Z5.61 : sync_plan_position``

still investigating...

I know i have incredibly bad differences, between readings, but they are consistent and probe never really goes above 50° anyway

```
Final compensation values:
Probe temp: 30.00C; Offset: 0.00 um
Probe temp: 35.00C; Offset: 134.00 um
Probe temp: 40.00C; Offset: 498.00 um
Probe temp: 45.00C; Offset: 589.00 um
Probe temp: 50.00C; Offset: 935.00 um
Probe temp: 55.00C; Offset: 1282.00 um
Probe temp: 60.00C; Offset: 1628.00 um
Probe temp: 65.00C; Offset: 1974.00 um
Probe temp: 70.00C; Offset: 2320.00 um
Probe temp: 75.00C; Offset: 2667.00 um
Probe temp: 80.00C; Offset: 3013.00 um
```

For my purposes I can set the intervals smaller, e.g. to 3 to have less measurements, then I am fine, but just seeing something going out of range there

closed 2020-07-01 by github-actions[bot]