MarlinFirmware/Marlin #16786: [BUG] TMC CURRENT changing when printing

Bansh1990, 2020-02-05:
**TMC CURRENT CHANGING WHEN PRINTING**
I have a missing steps on X axis, and I noticed something...
During printing, if i go to the TMC CURRENT menu, the value in the X axis changes.
this is normal?

 My Configuration

**ENDER 3 pro**
**TMC2208**
**SKR E3 DIP**

boelle, 2020-02-05:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Bansh1990, 2020-02-05:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4159904/Marlin.zip)
there is. thanks

> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

teemuatlut, 2020-02-06:
https://github.com/MarlinFirmware/Marlin/blob/54aff831791e50a1cfb4cf7ccef36f5fefff5494/Marlin/Configuration_adv.h#L2233-L2244

boelle, 2020-02-08:
@teemuatlut do you think this is a bug or just a misunderstanding?

ie can we close this one?

oliver-eifler, 2020-02-08:
in your configuration_adv.h the x_current is set far to high for ender 3 x/y/z stock steppers... lower it to the same value as y_current
#define X_CURRENT       530        // (mA) RMS current. Multiply by 1.414 for peak current.
and make sure drivers/board a cooled properly

closed 2020-02-08 by Bansh1990