MarlinFirmware/Marlin #16787: [FR] support st7735 (uclib) (ezt t1 board)

whoim2, 2020-02-05:
Sinis / Ezt / T1 delta printer have st7735 120*160 color display SPI. U8Cglib not support him, but UCglib from identical author - support. Libs have many identical things.
Please, add support for st7735 display. I may test it.
PS: chip stm32f103rbt6 need replace to stm32f103rct6 or ret6, its no problem.

Sorry, if my english is bad)

ugers, 2020-02-17:
I have exactly such plans. replace the chip and use marlin. I join the request to add a display.

whoim2, 2020-02-17:
> I have exactly such plans. replace the chip and use marlin. I join the request to add a display.

Go to telegram channel 
https://t.me/EZT_T1 (russian), we have many interesting )

majekw, 2020-05-20:
There is similar work for other graphic lcds not supported in u8g here: https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/lcd/dogm/u8g_dev_tft_320x240_upscale_from_128x64.cpp

closed 2020-07-01 by github-actions[bot]