MarlinFirmware/Marlin #16788: [BUG] Bricked SKR Mini 1.2 E3 Board after flashing

skaman82, 2020-02-05:
### Bug Description

I just received my SKR Mini E3 1.2 board. I tested it with Pronterface, and it was working fine.
I then downloaded the 2.0.3 Marlin release and sample config files (unchanged) for this board and installed VisualStudio and PlatformioIDE (Mac OS). It is building in VS, but after I flash the board, the board is unresponsive (not even showing up in my USB devices). Tested "STM32F103RC_bigtree" and "STM32F103RC_bigtree_512K" target. Any help appreciated.

### My Configurations

### Steps to Reproduce
see above. 
VS Version: 1.41.1 and PlatformIO 3.0.1 Core 4.1.0 (all latest versions)

**Expected behavior:** [What you expect to happen]
It should boot normally

**Actual behavior:** [What actually happens]
It is unresponsive

disconn3ct, 2020-02-05:
I may have the same problem. Does the board respond to the console? (Is the problem purely USB or is it the whole board?)

Is it bricked or can you reflash working firmware?

jmz52, 2020-02-06:
Please elaborate the flashing process. Did you update firmware from SD card using vendor's bootloader or via ST-Link?
Also please attach environment settings from platformio.ini

skaman82, 2020-02-06:
@disconn3ct yes, I can reflash the board with .bin files from the vendor and it is working again and is showing up as Mappe in my USB devices.

@jmz52 Flashing process: compile the .bin file, transfer to the SD card, flash the board with that SD card.

Environment settings: 
[platformio.ini.zip](https://github.com/MarlinFirmware/Marlin/files/4165824/platformio.ini.zip)

disconn3ct, 2020-02-06:
Definitely not bricked since it does stuff. Might update the title to something like "Printer hang" instead of brick - brick is effectively unrecoverable. (Usually recovery involves soldering or other extreme methods, not just reflashing good firmware.)

Try turning off USB_COMPOSITE (flash the non-USB version) and see if it comes back.

damvcoool, 2020-02-08:
@skaman82 
maybe try the modified configuration from my Repo, This has worked for me and a few of my local friends.
https://github.com/damvcoool/Marlin-2.0.x-SKR-Mini-E3-v1.2

RuairiSpain, 2020-02-10:
> @disconn3ct yes, I can reflash the board with .bin files from the vendor and it is working again and is showing up as Mapple in my USB devices. I can connect via Pronterface. But not with my compiled file - with that it is not showing up as a device in my computer.
> 
> @jmz52 Flashing process: compile the .bin file, transfer to the SD card, flash the board with that SD card.
> 
> Environment settings:
> [platformio.ini.zip](https://github.com/MarlinFirmware/Marlin/files/4165824/platformio.ini.zip)

I've checked your config files and they look OK.  Except, the is potential issue.  In the platform.io on line 827, you have:
`[env:rumba32_mks]` but for the 2.0.3 branch I think this should be: `[env:rumba32]`  

I don't have the same branch as you, so take anything I say with a grain of salt!

skaman82, 2020-02-13:
I actually solved it (somehow). I reinstalled the PlatformIO Core (downgraded it first). Then took a clean release files and started my config over. It compiled and after flashing the board it is working like it should :)

closed 2020-02-14 by thinkyhead