MarlinFirmware/Marlin #16791: SKR Pro V1.1 Laser Feature Enable Compile Error

Danielstc91, 2020-02-06:
Getting a repeatable error when compiling Release- Marlin 2.0.3 (as well as 2.0.x) for SKR Pro. Enabling laser feature produces compile errors even after defining required pins in ..stm32/pins_BTT_SKR_PRO_V1_1.h

Information
BigTreeTech SKR Pro V1.1 Board 

#define LASER_FEATURE uncommented in configuration.h

stm32/pins_BTT_SKR_PRO_V1_1.h 
added following lines
#define SPINDLE_LASER_PWM_PIN PC9
#define SPINDLE_LASER_ENA_PIN PF8

### Steps to Reproduce

1. Downloaded Release 2.0.3
2. Configure and Compile Successfully With laser feature disabled
3. Enable Laser feature in configuration.h
4. Define pins in configuration_adv.h
5. Compile

**Expected behavior:** Code Compiles Successfully

**Actual behavior:**
Commenting Laser feature out allows code to compile successfully.
Mentions and issue with FastIO in error messages.

#### Additional Information

Exact Error below:

[Marlin-bugfix-2.0.x - Laser attempts.zip](https://github.com/MarlinFirmware/Marlin/files/4162949/Marlin-bugfix-2.0.x.-.Laser.attempts.zip)

*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\Servo.cpp.o] Error 1
In file included from C:\Users\Todd_\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:48:0,
                 from Marlin\src\HAL\HAL_STM32\SoftwareSerial.h:36,
                 from Marlin\src\HAL\HAL_STM32\SoftwareSerial.cpp:39:
C:\Users\Todd_\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:254:51: error: missing binary operator before token "("
 #define digitalPinHasPWM(p)         (pin_in_pinmap(digitalPinToPinName(p), PinMap_PWM))
                                                   ^
Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/fastio.h:83:33: note: in expansion of macro 'digitalPinHasPWM'
 #define PWM_PIN(P)              digitalPinHasPWM(P)
                                 ^~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:2501:12: note: in expansion of macro 'PWM_PIN'
     #elif !PWM_PIN(SPINDLE_LASER_PWM_PIN)
            ^~~~~~~
In file included from C:\Users\Todd_\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:48:0,
                 from Marlin\src\HAL\HAL_STM32\../shared/Marduino.h:36,
                 from Marlin\src\HAL\HAL_STM32\HAL.h:28,
                 from Marlin\src\HAL\HAL_STM32\timers.cpp:25:
C:\Users\Todd_\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:254:51: error: missing binary operator before token "("

boelle, 2020-02-06:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Danielstc91, 2020-02-07:
My bad I thought having entire marlin folder would have been helpful as well. Here is the requested file

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4168470/Configs.zip)

boelle, 2020-02-08:
@Danielstc91 no we only need the config, we have the marlin folder allready :-)

sjasonsmith, 2020-02-23:
The `PWM_PIN` sanity checks don't work on STM32. The only real option right now is to comment out the sanity check, and verify for yourself that the pin has PWM enabled. It is disabled on a lot of pins, because using PWM on random pins often ends up messing up timers for things like stepper movement and heaters.

To check those, you need to look at the PWM table in the board variant:
`buildroot\share\PlatformIO\variants\BIGTREE_SKR_PRO_1v1\PeripheralPins.c`

There are other open issues related to this, but I'm not sure where they are at the moment.

sjasonsmith, 2020-02-23:
If anyone else want to look at this, here are some headers to reproduce it. The configs already attached aren't very usable in the current bugfix. These are made from scratch, not from the posted configs at all.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4240623/Marlin.zip)

FNeo31, 2020-02-23:
@Danielstc91  just go here: Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/fastio.h

find this:
 #define PWM_PIN(P)              digitalPinHasPWM(P)

change to this:
 #define PWM_PIN(P)             true

need to do this every time that I download a new version of Marlin -.-
Welcome to SKR Pro v1.1 World :)

Danielstc91, 2020-02-25:
Hey i will Test out next chance i get! Thanks for the help and will post updates within a couple days

thinkyhead, 2020-02-26:
Anyone want to have a crack at writing a proper `digitalPinHasPWM` implementation for STM32??

sjasonsmith, 2020-02-26:
> Anyone want to have a crack at writing a proper `digitalPinHasPWM` implementation for STM32??

Solutions I can imagine involve creating a pre-build script that would parse the peripheralPins.h file and produce a new header that provides constants for each pin.

Haschtl, 2020-05-06:
I'm not really a C or C++ programmer, but isn't the digitalPinHasPWM function defined in the Arduino STM32 Core repository?
There is at least the file [pins_arduino.h](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/pins_arduino.h), where `digitalPinHasPWM` is defined like this:
 ```
#define digitalPinHasPWM(p)         (pin_in_pinmap(digitalPinToPinName(p), PinMap_PWM))
```
Maybe this helps someone ... would be nice, if this gets fixed!

mjscosta, 2020-05-31:
> 
> 
> Anyone want to have a crack at writing a proper `digitalPinHasPWM` implementation for STM32??

I'm going to pickup this.

So the issue is caused by:

```
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2845:12: note: in expansion of macro 'PWM_PIN'
 2845 |     #elif !PWM_PIN(SPINDLE_LASER_PWM_PIN)

```
The PWM_PIN(P) macro is being defined using functions (as refered above). The digitalPinHasPWM(P) uses which is only defined at compile time.

Hence #elif !PWM_PIN(P) cannot be used here.

Possible solution is to use a set of constants/defines for PWM_PIN(P) or digitalPinHasPWM(P), maybe using some X-Macros "magic" to avoid repeated code.

mjscosta, 2020-05-31:
STM32F1 uses PIN_MAP in the PWM_PIN(P) define, but I could not find this external variable ... also not sure if it makes sense to use it ...

boelle, 2020-06-21:
@Danielstc91 still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

mgros, 2020-06-30:
There ia a solution for the problem?
I want to configure a Laser in SKR PRO V1.1

FNeo31, 2020-06-30:
Yes is not the best but works, check my comment from 23Feb. My machine are running with SKR Pro and have spindle and laser on it. And work fine

sjasonsmith, 2020-07-06:
A change just went in that bypasses the PWM sanity check on SKR Pro and other STM32 boards. It currently simply isn't possible to check that at compile time.

From re-reading this, that seemed to be the important bit for all of this. If there are remaining issues that need addresses please reply and let me know. We can re-open the issue if there was more to this.