MarlinFirmware/Marlin #16795: [BUG] Fysetc S6 emulated eeprom issue bl touch wont pull out the pin after M500

TB1405, 2020-02-06:
i confirmed this issue on the Marlin 2.0.3 and actual bugfixes2.0.x

### Bug Description
if i change a value that will be safed in eeprom the homing doesnt work. 
problem is the bltouch doesnt then deploy the pin out before homing Z axis. 
but when i restart the printer the works normal including bl touch. 
if i change now again a eeprom value and safe it the homing wont work again (BL touch pin wont push out) 
my marlin file is attached. 

### My Configurations

[MarlinS6-EEPROMProblem-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4166809/MarlinS6-EEPROMProblem-2.0.x.zip)

### Steps to Reproduce

1. download my config and compile it in VSC 
2. load it to fysetc S6 
3. change a value (accel or babystep or homingthreshold etc.) and safe it on eeprom ( i know it isnt a real eeprom)
4. click the auto home button

**Expected behavior:** succesful homing 

**Actual behavior:** the homing for  x and y axis is without issue. when homing z the bltouch wont push out the pin and the printer will ram in the bed 

restart the printer now and click on homing again but dont change or safe any value on eeprom before
**Expected behavior:** succesful homing 

**Actual behavior:** succesful homing 

#### Additional Information
iam using in this configuration a ftdi touch but should be the same problem with other displays.

neo799, 2020-02-07:
I got the same issue. Printing works fine but as soon I use M500 the BLTouch won't work anymore for Homing etc.

BastR, 2020-02-10:
I had that problem, then it disappeared BUT I have that problem again too.

On SKR PRO, I had opened an issue here https://github.com/MarlinFirmware/Marlin/issues/16169.

I may be wrong, but I seem to remember it was about the TIMERS.

@sjasonsmith Seems to me you've been working on that? (sorry if I'm wrong)

TB1405, 2020-02-12:
I activated the i2c eeprom but doesnt have one and it works correct i dont know why but if you activate i2c eeprom Flash eeprom works correct

GhostlyCrowd, 2020-04-22:
Just wanted to Post that this is still an issue on Buxfix 2.0.x branch on the SKR Pro 1.1 Exact same issue any eeprom saving causes tool head crash because of failure to deploy probe.

randellhodges, 2020-04-22:
> Just wanted to Post that this is still an issue on Buxfix 2.0.x branch on the SKR Pro 1.1 Exact same issue any eeprom saving causes tool head crash because of failure to deploy probe.
> 
> SD Eeprom Emulation is a work around that fixes this

If you can, try the following:

1) Go back to flash based eeprom
2) Verify your bltouch is working and you can see the little blue led on the side of the bltouch
3) Issue an EEPROM save
4) Don't home or anything, just look and see if the bltouch still has the little blue led on
5) If it is still on, issue the servo commands to drop/raise the pin
6) If that works, then see if homing works

Might help narrow some things down

trouch, 2020-05-11:
Should be fixed with https://github.com/MarlinFirmware/Marlin/pull/17946

closed 2020-06-20 by boelle