MarlinFirmware/Marlin #16797: X_MAX_PIN redefinition breaks Anycubic Kossel Linear Plus

n0n3m4, 2020-02-06:
### Bug Description

In the file pins_TRIGORILLA_14.h there is a redefinition `#define X_MAX_PIN          43` that breaks x_max endstop on Anycubic Kossel Linear Plus. I don't really know why it is there (so I'm not providing a pull request), but I commented it out and my endstops are working again.

### My Configurations

I'm using Anycubic Trigorilla 1.1, I think this is quite important. Configuration files are of no help here, though.

### Steps to Reproduce

Flash firmware and press autohome without commenting this line. Screeching sound will let you know that something is going very wrong. `M119` reveals that x_max is always triggered.

thisiskeithb, 2020-02-06:
Can you attach your configs & modified pins file? `X_MAX_PIN` should only set to 43 if you define `ANYCUBIC_4_MAX_PRO_ENDSTOPS`.

https://github.com/MarlinFirmware/Marlin/blob/3a3429b1ef2fdaf1e793fff17a89b069e3384fbc/Marlin/src/pins/ramps/pins_TRIGORILLA_14.h#L40-L47

sjasonsmith, 2020-02-06:
I assume this is with 2.0.3? That has been fixed already in bugfix-2.0.x. There was some confusion regarding pin assignments because Anycubic uses different pin numbers for endstops on the ANYCUBIC_4_MAX_PRO, even though it uses the same motherboard.

thisiskeithb, 2020-02-06:
This is why we ask for configs. 🙂

sjasonsmith, 2020-02-06:
Here is the original issue reporting this: #16736

n0n3m4, 2020-02-07:
Wow, yes, its 2.0.3. I guess I'm closing this then, thanks, sorry for duplicate bug report.

closed 2020-02-07 by n0n3m4